FreeCAD/FreeCAD #5538: Basic Assembly stuff

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=16

* **Reporter:** Jriegel
* **Date submitted:** 9/28/2009
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

Basic function in the assembly module like:

Moving parts around
simple constraints like:
* contact
* axle
* offset

There for we need the Navigator and usage of the ODE lib

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by mntc 2013-10-01 18:15
May be assembly constraints can be:

a) scriptable (One can rewrite standard primitive constraints by complex python script determining behaviour of constrained objects)
b) hierarhical (In Autodesk Inventor parts in subassemblies are frozen when we open assembly of higher order. It is sometimes useful, sometimes not. It could be good to switch between styles of assembly behaviour inside of a superassembly.)
c) built into special objects (may be instead of adding constraints, we should add special objects - bearings, etc. E.g. like in Algodoo, where constraints are determined by nature of objects.)

---

### Comment by Pauvres_honteux 2014-05-18 07:26
Hi Jürgen, would it be too bold to ask for a more elaborate explanation/overview of what you want to accomplish with this assembly feature?
I got a picture in my head, when I read the headline, of something similar to what you have in the big french program. Or do you have bolder ideas in mind?

Regards Pauvres honteux.

---

### Comment by Kunda1 2018-06-13 13:28
Unassigning Jriegel

donovaly, 2022-06-24:
The Assembly module was removed and does no longer exist.

closed 2022-06-24 by donovaly

luzpaz, 2022-06-24:
The point of keeping this ticket was to close it when Assembly WB is finished. But nm

donovaly, 2022-06-24:
As I wrote, the remaining traces of the unfinished Assembly WB were removed before the release of FC 0.20.

luzpaz, 2022-06-24:
Yes, I understand that it relates to the Assembly WB that is deprecated. But it really relates to a functioning Assembly workbench in general (which we don't have yet)  I'm not sure you understood, It was more symbolic ticket. I would of closed it a long time ago. But it was ticket number 00016 in the old mantis tracker.