FreeCAD/FreeCAD #5540: Python debugger

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=69

* **Reporter:** wmayer
* **Date submitted:** 1/29/2010
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Add a Python debugging facility to FreeCAD.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wmayer 2012-12-29 16:37
External resources:
http://hapdebugger.sourceforge.net/

http://pydev.org/download.html
http://pydev.org/manual_101_root.html
http://pydev.org/manual_adv_remote_debugger.html

http://winpdb.org/
http://winpdb.org/docs/embedded-debugging/

---

### Comment by wmayer 2013-01-09 17:42
See also:
QScriptEngineDebugger
QScriptEngine
QScriptEngineAgent

https://github.com/hpfem/agros2d/commit/e92e5b8f8fef3b9eb13e81a9d79d21139a36b161
https://github.com/hpfem/agros2d/blob/master/src/pythonlab/pythonengine.h

---

### Comment by Kunda1 2017-04-28 19:41
Proposed Python editor/debugger workbench: https://forum.freecadweb.org/viewtopic.php?f=10&t=21898

---

### Comment by Kunda1 2017-07-12 18:29
Updates to debugger: https://forum.freecadweb.org/viewtopic.php?f=10&t=21898&start=10#p182177

luzpaz, 2022-03-13:
CC @mumme74 (JIC you were interested in following the developments)

dprojects, 2022-03-13:
Hello, Luzpaz told me about it here: https://github.com/dprojects/scanObjects/issues/1 
Where and how to submit the code? you can do fork? and maintain it?

WandererFan, 2023-10-23:
Closed as out of scope.