FreeCAD/FreeCAD #5544: add support for Rhino3D files

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=337

* **Reporter:** e-ndy
* **Date submitted:** 4/14/2011
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** file format, NURBS

# Original report text

available library from developers of Rhino3D (http://opennurbs.org/). it should be nice to have capability to load/save Rhino3D models, together with https://sourceforge.net/projects/libnurbs/ it should be good replacement for Rhino3D in linux

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Jriegel 2011-05-17 07:35
Actually I dont like myself to struggle with commercial file formats. But If someone have coding time - be welcome.

---

### Comment by normandc 2011-05-21 11:51
It is unclear to me reading the OpenNURBS Initiative website whether the 3DM file format is open source or commercial. But it seems to be completely documented, which I think means it would be much less of a struggle to incorporate in FreeCAD as other proprietary and closed formats. It would indeed be really nice to have a CAD program on Linux able to open 3DM files. But there is no NURBS support or tools in FreeCAD at the moment, maybe Blender (which supports NURBS surfaces) would be best suited?

---

### Comment by Jriegel 2011-05-22 18:25
OCC support NURBS, but we lack the tools to model it. 
So we could do boolean operation on loaded NURBS, buts 
thats around it. 

Actually I'm not a big fan of NURBS. IMO subdevision surfaces are 
the way of the future...

---

### Comment by yorik 2011-05-29 13:54
Actually since freecad supports nurbs, 3DM format should be possible to handle without problems (and IIRC Rhino also supports brep-like objects)... On the openNurbs website there is a C++ toolkit which can read/write 3DM files, and appears to be compatible with GPL license ( http://www.opennurbs.org/docs.htm ):

"Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files (the "Software"), to deal in the Software without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of the Software."

I'm downloadiing the toolkit right now to have a look...

---

### Comment by yorik 2011-05-29 13:59
Interesting quotes from the FAQ included inside the toolkit:

"Excellent compilers and make utilities are available from
The Free Software Foundation <http://www.fsf.org>.  You have
the option of using these tools free of charge."

"Internally, Rhino stores all surfaces as some type of b-rep and the
openNURBS toolkit reads these objects as b-reps."

---

### Comment by yorik 2011-05-29 14:04
Hm reading a file looks actually pretty simple, if we can include the opennurbs code!

---

### Comment by unauthenticated 2011-06-01 14:30
Reading the file should not be the problem.
You have to transfer each entity to a OCC object.
Not only the NURBS, also the trimming curves and all
other entities like planes and so on... 

Its posible, but a lot of work an tuning to get it run....

---

### Comment by keithsloan52 2014-08-02 23:06
I have a skeleton FreeCAD importer that is using the OpenNurbs library to parse a 3dm file. I will now try and add code to create nurbs etc

---

### Comment by Kunda1 2017-01-11 12:56
Picking up the trail here...a forum thread mentioning this topic: 
<b>Nurbs Editor</b> - http://forum.freecadweb.org/viewtopic.php?f=3&t=16473&start=30#p132955 
<b>Import 3DM files with OpenNurbs</b> - https://forum.freecadweb.org/viewtopic.php?f=3&t=16473&start=130#p153127
GitHub repo where the development is happening; https://github.com/KeithSloan/FreeCAD_ImportNurbs

---

### Comment by Kunda1 2017-04-26 00:52
travm1 wrote:
**Here is the [wiki](https://www.everipedia.com/Rhinoceros_3D) page for Rhino3D**

---

### Comment by Kunda1 2017-04-26 00:55
keithsloan52 wrote:
Yorik has done some work on using CAD Exchanger see https://github.com/yorikvanhavre/CADExchanger So should be able to import Rhino 3DM files using it. Note CAD Exchanger is commercial so one needs to pay. Not sure how much as you seem to have to contact them to get a quote.

---

### Comment by Kunda1 2017-04-26 00:57
yorik wrote:
**They told me they are setting up a new page that will have prices on it. In any case, implementing our own might still be an interesting option.**

---

### Comment by thalamus (private) 2017-10-20 18:26
awesome, thanks travm
[Thalamus](http://www.thalamus.co)

kkremitzki this is spam ^
can we ban this user please?

---

### Comment by Kunda1 2020-03-23 12:10
Update: Chris_G on the forum utilizing rhino3dm python bindings:
https://forum.freecadweb.org/viewtopic.php?f=8&t=11970&p=379662#p379662

---

### Comment by keithsloan52 2020-03-24 16:46
A new Workbench has been started using Chris's code and use of the new rhino3dm python bindings see https://github.com/KeithSloan/ImportNURBS

luzpaz, 2022-03-11:
CC @KeithSloan

KeithSloan, 2022-03-11:
Probably not fully but my ImportNURBS workbench can read Rhino3D files. An inhibitor to further development is that you cannot do much with them other than display them. I am not aware of any of workbenches for dealing with NURBS allowing for editing, they all just allow you to create, each with their individual methods of creation. I raised an issue with the Curves Workbench see  https://github.com/tomate44/CurvesWB/issues/59.

Not sure but I think some people have imported 3DM files and then used the Path Workbench (this is unconfirmed).
If there was the facility to edit Nurbs then there would be an incentive to add export facilities.

Maybe a GSOC project but would need Chris_G or somebody to mentor.

maxwxyz, 2024-02-25:
Is this still a relevant request?

KeithSloan, 2024-02-25:
Well the ImportNurbs Workbench https://github.com/KeithSloan/ImportNURBS has 10 stars and there were 28 clones in the last seven days. github users that have stared sometime in the past are
[image]

But at the end of the day what is required is the ability to edit and alter the imported curves and surfaces, maybe a future curves type workbench?

Then is would be worth while adding support for export of Nurbs.

There is no Open Source equivalent of Rhino and if the ability to edited such curves and surfaces was created then FreeCAD would acquire quite a number of users.

yorikvanhavre, 2024-02-26:
An additional note, the openNURBS library, which is used to import/export Rhino files, is GPL licensed. So it will likely never be a core feature of FreeCAD because we don't want pure GPL code in, but rather an add-on.

Long time I didn't play with your importer @KeithSloan ... Should give it a new test spin

KeithSloan, 2024-03-09:
Rhino allows modelling through creation of Nurb curves and surfaces for some incite of what this enables see https://mcneelmiami.hflip.co/4ae71071b2.html

KeithSloan, 2024-03-23:
Made a start on new exporter Module [https://github.com/KeithSloan/ImportExport_3DM](url)

see [https://forum.freecad.org/viewtopic.php?t=22675&start=1820#p749032](url)

It is a start Sweep_Rotation (BSpline curve )from file LogSnail_4 exported tp 3DM and opened in Rhino
[image]

KeithSloan, 2024-12-07:
I just tried to install the rhino3dm library in FreeCAD 1.0.0 and whilst it works in native python3.

Under FreeCAD I get

File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/ImportNURBS/./freecad/importNURBS/import3DM.py", line 38, in <module>
    import rhino3dm as r3
  File "/Applications/FreeCAD 1.0.0.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD 1.0.0.app/Contents/Resources/lib/python3.11/site-packages/rhino3dm/__init__.py", line 5, in <module>
    from ._rhino3dm import *
  File "/Applications/FreeCAD 1.0.0.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'rhino3dm._rhino3dm'