FreeCAD/FreeCAD #5545: Switch on and off parametric behaviour of document objects

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=402

* **Reporter:** wmayer
* **Date submitted:** 7/18/2011
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

According to the thread [1] we need a way to switch on and off the parametric of document objects. Since this should be serialized we have to use a PropertyBool member. If this flags is off we simply have to skip the execute() method inside the recompute() of DocumentObject.

[1] http://forum.freecadweb.org/viewtopic.php?f=8&t=1070&p=8654#p8652

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by bejant 2015-05-28 14:01
Here's a new link to the topic, after the FreeCAD forum moved from SourceForge:
http://forum.freecadweb.org/viewtopic.php??f=8&t=1070&p=8654#p8652

---

### Comment by Kunda1 2018-06-13 13:27
Unassigning Jriegel

---

### Comment by Kunda1 2018-08-26 11:20
Updated URL in ticket summary from old sourceforge URL

---

### Comment by Kunda1 2019-01-01 13:13
New discussion at https://forum.freecadweb.org/viewtopic.php?f=8&t=33101

adrianinsaval, 2023-04-02:
I this relevant? I think this use case is now covered well by the simple copy command, consider closing this issue

closed 2023-04-02 by chennes

FlachyJoe, 2024-02-23:
@wwmayer you tell me about this issue, do you think a Freezed state is needed for DocumentObject in order to prevent re-computation?
I had a look on the code and I think it's easy to implement.
A freezed object will always return false to `mustRecompute()` and it's properties will be readonly.

wwmayer, 2024-02-24:
> you tell me about this issue, do you think a Freezed state is needed for DocumentObject in order to prevent re-computation?

Unlike to my suggestion at that time I wouldn't use a PropertyBool any more because this has an extra cost of additional memory for every object type which is unused in 99% of all cases. I prefer to extend the ObjectStatus enum with a further value e.g. "NonParametric" or "Frozen" or whatever.

> A frozen object will always return false to mustRecompute()

Yes.

> and it's properties will be readonly

This will be a bit tricky because the flag can be reverted and there is no easy way to restore the original status of all properties. I would just leave the properties untouched. It should be checked if doing nothing inside `DocumentObject::onChanged` is sufficient.