FreeCAD/FreeCAD #5548: Inconsistent translations for workbench names

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=495

* **Reporter:** normandc
* **Date submitted:** 11/27/2011
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** locale, translation

# Original report text

Currently in latest FreeCAD source, translation for workbench names is inconsistent:

- Some workbenches toolbar names are translated (in View --> Toolbars menu)
- Some workbench specific menu names are translated, some aren't. For example, when the workbench selector is set to a specific workbench, the workbench menu name is not translated. The Raytracing menu name is in English (but its contents are translated). But if in complete workbench mode, then most of the workbench menus are translated.
- In the workbench selector, the workbench names are not translated.

This should require some consideration.

I realize that translating the workbench names is a delicate matter as it could complicate communication on the forums with users of other languages. But I believe that a wholly translated interface is preferable, as some English names don't have signification to users of other languages. (For example: the distinction between Draft and Drawing can be lost on French users)

# Additional information

I'll try to post screenshots shortly.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Bug
* **Updated:** 3/2/2021

# Discussion from Mantis ticket

### Comment by Jriegel 2011-11-27 18:00
We have different levels of translations. Thats IMO not a bug. 
Besides there is a string not showing up on crowdin, that would be a bug!

Otherwise its a encouragement for the translaters and I will close this "bug" ;)

---

### Comment by normandc 2011-11-27 18:16
Please explain what you mean by different levels of translation.

Right now, I have workbench menu names showing either in French or in English depending on if I've set the complete workbench, or if the workbench selector is set to a specific workbench. Then the workbench selector itself is not translated. How is that no a bug? And why isn't a single string used for the 3 use cases?

---

### Comment by yorik 2011-11-27 19:12
I think workbench names are specified in their InitGui file, which is why they are not taken into account by the translation system. Since workbenches are activated/deactivated by name internally, changing this will require serious changes in the code. Sincerely I'm not sure it's worth the work...

Their contents (toolbars, menu entries, etc) on the other hand, is taken into account by translation mechanisms and should be 100% translated. If something is missing (provided all crowdin strings are translated of course), indeed it is a bug (probably some string internally which isn't using a translation function). I suggest you make a list here with what you see that isn't translated, we'll need to go one by one...

But I'm not sure it's a showstopper for 0.12... There will probably be more than a couple, and at this rate that release will never be out ;) My opinion is more to see translations as a work in progress, and refine progressively...

---

### Comment by normandc 2011-11-27 19:43
What I want to make sure of is to get a sense of the direction we go. If we decide that workbench names should not (or cannot) be translated, then I will have to revert all the translations where I have translated them. And that includes the Wiki documentation...

So that to prevent inconsistencies in v0.12...

---

### Comment by normandc 2011-11-29 02:21
I finally made screen captures.

I used the Draft module as an example.

[... 100 lines cut ...]
I realize that they are "small" disadvantages, but they increase "confusion"... Even today several FreeCAD users complain about the confusion, some love FreeCAD, but they say it anyway... so we should avoid adding more, even if minimal.

I agree that this "confusion" often does not exist, and it is only the user who remains attached to the old experiences, or maybe is only used to the "ready meals" of commercial software, but nothing can be done about this.

I can change my mind about anything I've written if it's wrong, I hate beliefs and love the benefit of the doubt, so if something escapes me I'll be happy to understand it, thank you.

I apologize for the bad English, written with the help of google translated.
A greeting and thanks for all the work done to date, you are great!

---

### Comment by MaurinoWeb 2021-03-02 23:17
An example that concerns only the SEO with the changes of WB names.
Recently the name of the part design workbench has been translated into Italian in "ambiente disegnopezzo"

I find it smart to translate "workbench" but the rest(wb names) should be a fixed word, so the user has the help of the translated "workbench" to understand choise what is, but the WB names is part of the FreeCAD fixed terminology and will soon learn.

Now looking at html of the view-source:https://wiki.freecadweb.org/PartDesign_Workbench/it
you can see that in the <title> tag there is no term as in the url, this is not recommended.
it is also not recommended to change the <title> since it is an official documentation, now if we write in google "ambiente part design" in the "search result" still has the "old" title (ambiente part design) while duckduckgo has already been updated (ambiente disegnopezzo), soon google will update too.
We are not helping the search engines, so one day it could better position a site with a seo-friendly html and find the official documentation under the results, in English no, but in other languages ​​it could happen and these changes do not help.

So imagining all the WB names translated, I confirm my concern that in all languages ​​other than English there will be scattered rankings with different "key phrases" in search engines... this won't help over time.

a greeting

luzpaz, 2022-05-09:
CC @kaktusus @Kuzma30

kaktusus, 2022-05-14:
The initial assumptions are good.

Keep in mind that the topic may be considered individually by different nationalities.

Creating a generic mechanism to translate workbench names as well will never be a bad or wrong solution.
The users _(translators)_ will get a well-functioning mechanism to use.
Everyone can translate a name according to "his or her liking" _(or rather the user community of a given country)_.

This is important to remember.

The Polish translation team has decided that these names will also be translated into Polish.

However, I still come across users who do not like it, and prefer to switch to the English version. And no one will forbid them this. Mostly, these people are familiar with programming and have some habits they may not be aware of.

It has to be said that until some time FreeCAD could not enjoy a well and fully prepared interface in Polish language. I'm happy to say that this has changed.

There is still a lot of work ahead of the team of translators as a lot of attention will have to be paid to perfecting the translations. But there are already many users satisfied with the result.

I think that giving the users _(translators)_ a good tool in the form of the possibility to translate the names of the workbenches is necessary. However, one should not consider and look for a mistake in the way it will be used.

**It is the responsibility of the translation teams to decide whether and how selected GUI elements will be translated.**

_**But they must have the right tools and capabilities to do so.**_

chennes, 2023-05-05:
I'm going to bring this issue back up to see if we have achieved a sort of consensus here: my read of the situation is that
1. All workbench names in user-visible parts of the interface should be **translatable**
2. For each language, the translation team decides whether or not to actually translate the name, possibly on a workbench-by-workbench basis.

So the remaining work to do on this ticket is for developers to ensure that we are meeting point 1, and for translators to document whether or not they are translating the WB names in point 2. Does that seem correct?

kaktusus, 2023-05-05:
Some teams of translators have written down rules about this issue.
You can read about them here: https://wiki.freecad.org/WikiPages#Other_languages

chennes, 2023-10-23:
Where are we still missing translatable workbench name strings?

kaktusus, 2023-10-23:
Without thinking, I can answer:
workbench selector

[image]

kaktusus, 2023-10-23:
Preferences → Workbenches

[image]

markit, 2024-07-29:
I'm testing 0.22_dev 
OS: Ubuntu Oracular Oriole (development branch) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
and I find very confusing that in the Workbench selector I've "Draft" (is not translated, but I think is ok, is easy to learn what it does and means), but as soon I go to the "Preferences" I have "Draft" translated into "Sformo".
Here 2 problems: 1) "Sformo" means nothing in italian 2) I should find the same name of the workbench here, so I know that workbench preferences I'm working on.
So please revert back to previous 0.21 behaviour to avoid confusion with this flip/flop "somewhere" translation

dbdbdb, 2024-07-30:
Yes. Found this issue when using Swedish and the current dev version (0.22) and it's totally confusing.

The workbench is called "Draft" but in the preferences it's called "Släppning" which makes it very hard to do the connection between the two. The translation is actually for the draft tool in the part design workbench and not for the draft workbench but even if the translation is corrected it would still be difficult. We should refer to the workbench using the same name in both the workbench selector and in the preferences.

I can live with the workbench being translated or not. It just has to be consistent.

FEA-eng, 2024-08-28:
Any chance of having it fixed for 1.0? The workbenches list is very important and gives a bad impression when not fully translated.

hasecilu, 2024-08-28:
On Spanish we have reverted some legacy translations where WB names where oddly translated on preferences page, if I'm not wrong core WB names on ComboBox or on "Workbench" preferences (above pictures) **were never showed translated** (which is fine, choice was an illusion), so in order to be consistent we don't have to translate them on their individual preferences pages, thus the need of fixes.

The only WB that got showed translated were _external Python WB_, in that case in the FreeCAD addons project on CrowdIn names should be restored if they were translated to maintain the same rule.

The last translation update will bring latest changes that will let us find more errors, is there any other thing we could do?

kazan417, 2026-09-07:
Is there any progress ? As far as i know the decision was that translation team shoud decide to translate workbenches names or not. But looks like combobox is not showing translated version while menu is showing translated version.