FreeCAD/FreeCAD #5549: Measurement framework

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=503

* **Reporter:** wmayer
* **Date submitted:** 11/30/2011
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=1965

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Jriegel 2012-08-23 09:19
We need a bit more for a "Framework" :) 
I would put that in the code snipped of the Wiki...

---

### Comment by wmayer 2013-02-11 13:40
http://forum.freecadweb.org/viewtopic.php?f=10&t=3549&start=20

---

### Comment by wmayer 2013-07-05 07:26
http://forum.freecadweb.org/viewtopic.php?f=3&t=4225&start=10

---

### Comment by mrlukeparry 2014-01-08 20:12
The Drawing Module I am working on has the facility for measurements and basic statistics that could be extended. This is in a separate module Measure and has python bindings too.

---

### Comment by wmayer 2015-02-01 17:02
http://forum.freecadweb.org/viewtopic.php?f=8&t=9561&p=77531#p77531

---

### Comment by wmayer 2017-10-11 13:14
https://forum.freecadweb.org/viewtopic.php?f=22&t=24835

---

### Comment by Kunda1 2018-08-26 00:34
Hey easyw-fc does Manipulator WB address this ?

Edit: fixed old sourceforge phpBB url in the ticket summary to point to correct thread.

luzpaz, 2023-02-09:
@chennes this FR came up in Brussels 2023 right?

chennes, 2023-02-09:
Yes: lack of a unified and universally accessible measurement tool was widely agreed as a significant pain point. Measurement was viewed as deeply fundamental to CAD work, and the idea of having to switch WB, or even to install an external addon, to access it was... "unpopular."

WandererFan, 2023-02-09:
There is a Measurement WB.  It doesn't have a Gui, so not many people think about it.  It does have the ability to calculate measurements from geometry.  Don't remember if it has the ability to save the measurement across sessions. 

Luke Parry's note above references this.

A good starting point would be to merge the measurement tools from App? Gui? into this WB.

luzpaz, 2023-02-09:
CCing @easyw on this issue (because of the awesome work on Manipulator WB)

adrianinsaval, 2023-04-02:
I think we can close this one as superseded by #8561

chennes, 2023-04-02:
Yes, that issue is a good update for this one, I'm closing this Issue in favor of that one.

closed 2023-04-02 by chennes