FreeCAD/FreeCAD #5551: Macro editing FCMacro file - 4 improvements

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=573

* **Reporter:** pperisin
* **Date submitted:** 1/17/2012
* **FreeCAD version:** 0.12
* **Category:** Feature
* **Status:** new
* **Tags:** #lowhangingfruit, macros, split ticket

# Original report text

Hi,

I have three improvements suggestion that would help people in scripting

1 - auto indent - if the line I'm writting something in has i.e. 6 spaces before first character, when I press enter I go to the next line but at it's beginning. I would like to have 6 spaces automatically added when I press enter in this case.

2 - multi line editing - The easiest way to explain this one is if you open some document in notepad++. then press ALT key and select multiple lines. you will see how it selects in a row+column like way. Also, when you have stuff selected this way, writting anything will add those characters to all those selected lines.

3 - code completition and a list that appears after you press '.' - like the one in python console

4 -  Find and replace text 
Regards,
Petar

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by pperisin 2012-01-18 06:00
oh, another one - Find and replace text

---

### Comment by Kunda1 2017-02-02 14:15
Do you have any interest working on these features at all ?

---

### Comment by Kunda1 2017-04-29 12:54
Check out mumme work in https://forum.freecadweb.org/viewtopic.php?f=10&t=21898
(Also attached #69 to this ticket)

luzpaz, 2022-04-11:
CC @mwganson

luzpaz, 2022-05-26:
@mwganson can we close this ticket based on your macro?

mwganson, 2022-05-28:
I'm not opposed to it if somebody wants to work on it, but I don't have any plans at this time to work on adding any of these features.

luzpaz, 2022-06-07:
If there are devs interested, please check out https://github.com/mwganson/Editor_Assistant/issues/11

adrianinsaval, 2022-06-21:
IMO the best way to tackle this would have been using ktexteditor instead of the current, at least for indentation, search and replace, block editing and probably some code completion.

WandererFan, 2023-10-23:
Closed as not planned.

closed 2023-10-23 by WandererFan

mwganson, 2023-11-11:
This PR adds the automatic indentation when pressing Enter key.

https://github.com/FreeCAD/FreeCAD/pull/11356

Note: Ctrl+Space already provides an autocompletion mechanism.  I believe it is based on searching for text that already exists in the current document that could be used to automatically complete the current word.

luzpaz, 2023-11-12:
Thanks @mwganson! 
In regards to multiline editing, would that be something that could be implemented in the python editor ?

mwganson, 2023-11-12:
I don't have any plans for working on multiline editing at this time.