FreeCAD/FreeCAD #5554: Sketcher Fully constrained report based on primitives (Lines, arcs)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=704

* **Reporter:** pperisin
* **Date submitted:** 5/15/2012
* **FreeCAD version:** 
* **Category:** General
* **Status:** assigned
* **Tags:** constraints, Solver

# Original report text

HI,

I do not know if this makes sense, but I know it was mentioned before somewhere.

It would be nice that "fully constrained" report is based on single lines/arcs, and not on entire sketch. This way it would be easier to identify which parts of a sketch are not fully constrained.

Regards
Petar

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** General
* **OS: Win7 latest unstable **
* **Platform:** 32 bit
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by djreichel 2012-05-16 22:45
Maybe it would be useful to make lines display in a constrained color and an unconstrained color individually, rsther than making the entire sketch change to the constrained color when all degrees of freedom are removed.

It can be difficult to find the last unconstrained elements of a complex sketch.

---

### Comment by pperisin 2012-05-17 06:18
That is exactly what I ment. Would be very nice.

---

### Comment by pperisin 2012-06-05 12:23
Also, it would be nice if overconstrained primitives would be drawn in red (or some other color that is not used and that will mark over-constrained primitive)

EDIT: OR, even better, change color of conflicting constraints, so that user sees them on screen, selects one he does not need, delete it and be done with it.

maybe pink/purple color for conflicting constraints can be good. I'm bad with colors chosing. Really bad.

---

### Comment by Kunda1 2017-01-18 22:10
logari81 should this be tagged 'Solver'

---

### Comment by Kunda1 2017-06-16 11:51
trampas writes in #2653
[quote=trampas]What would be nice is to have lines change colors if they are fully constrained, or maybe have points change colors if they are matched.[/quote]

---

### Comment by Hannu 2017-06-27 15:40
This seems as a simple solution to a problem that is easy to run into in FreeCAD - I have used FreeCAD intensely in periods (creating objects to 3D-print) and still "hit" this, and I get as baffled every time.

---

### Comment by abdullah 2018-06-15 17:02
As for showing fully constrained elements, the problem is as in here:
https://freecadweb.org/tracker/view.php?id=2069#c11443

As for getting functionality to identify where the dossier is not yet constraint, the recently merged DoF Finder should help, just click on the hyperlink of the DoFs.

---

### Comment by abdullah 2018-10-07 11:05
As with #0002069, what is not implemented is constantly adapting the colors. This is not happening in v0.18, as complex sketches would have a significant drop in performance due to constraintly solving using the DenseQR algorithm.

adrianinsaval, 2022-03-10:
This has been implemented if my interpretation of the request is correct. Nowadays sketcher identifies and colors fully constrained individual geometries. This can be closed IMO.

abdullahtahiriyo, 2022-03-12:
Yes. Thanks. It is important to note that there are some bugs in which this functionality does not work well, but there are other tickets for that.

closed 2022-03-12 by abdullahtahiriyo