FreeCAD/FreeCAD #5557: [Sketcher] Dragging + snap to grid

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=847

* **Reporter:** pperisin
* **Date submitted:** 9/27/2012
* **FreeCAD version:** 0.12
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** #lowhangingfruit, #post-to-forum, snap

# Original report text

Hi,

Snap to grid works nicely in Sketcher, but there is problem when dragging a line. IN this case snap to grid only works if a mouse location is next to grid. It would be nice if line "Snaps" if one of it's endpoints is also near grid. 

Regards,
Petar

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by mrlukeparry 2014-02-08 00:23
The algorithm is pretty simple and you're right just uses the mouse cursor in relation to the grid. This is similar problem to dragging nodes to other edges and automatically adding constrains. This probably would need discussing with logari

A lot of improvements are needed to the grid in general such as being full screen. 

---

### Comment by Kunda1 2017-05-02 21:49
Unassigned

---

### Comment by Kunda1 2017-06-15 12:14
abdullah is this not implemented in sketcher 0.17 ?

---

### Comment by abdullah 2017-06-15 12:41
kunda1

I have to check, as I never use this function...

No, it is not. I just checked. Works nicely on creation, but on dragging it only enforces the grid if you are very close to one point. Just as the reporter describes. You may "low hanging fruit" this.

---

### Comment by abdullah 2018-11-01 18:40
https://github.com/FreeCAD/FreeCAD/pull/1767

---

### Comment by abdullah 2018-11-05 18:40
wwmayer commented 7 hours ago

I don't think it's a matter of preference, it's a matter of usability.

Setting such a huge tolerance doesn't make any sense because effectively it's not easily possible any more to move a point somewhere inside a grid cell. If you want to achieve it you must always go back to the task dialog and tmp. disable the option.

And if you start to draw a new line you click somewhere but don't get any feedback that the first point is already accepted and when pressing twice an error is raised that adding the line failed because both points are equal. In order to see the preview line you must move at least half a size of the grid cell along X or Y axis.

I think this odd behaviour will be confusing to many users.

wwmayer commented 7 hours ago

Btw, I looked at the feature request 847 and changing the snapping tolerance doesn't do what the OP wanted. If you have a line where none of its endpoint is snapped it still won't snap when starting to drag it.

abdullah:

Ok, it is a misunderstanding of what gridsnap should do. As I understood it, one point of having the gridsnap on may be precisely not to be able to move points inside the grid. Under such assumption, using 0.5 precisely prevents points outside the grid. When creating a new line, with an appropriate grid size, as soon as the line passes 50% of the grid size, you get a line segment of exactly one grid size measure. Of course using inappropriate values of the grid size leads to unusable behaviour. Because endpoints of lines cannot be outside the grid, when dragging points snap to the grid. Of course, if one creates a line without snap on, outside the grid, then sets snap on and drags, it won't align.

Of course, if weak snap, as currently implemented in FC is how it should work, this PR makes no sense.

Regarding the ticket:
Dragging an endpoint of the line works perfectly with weak snap. 

I am not sure I will implement dragging the edge and trying to snap the endpoints.

luzpaz, 2023-03-03:
@PaddleStroke is this ticket relevant to #8387 ?

PaddleStroke, 2023-03-03:
It's in the same thematic, but we did not worked on snap when dragging in
this PR. So we didn't changed current behavior.

But it is something that needs to be done.