FreeCAD/FreeCAD #5558: Write own and better version of BRepFeat_MakePrism

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=850

* **Reporter:** jrheinlaender
* **Date submitted:** 9/28/2012
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

This will resolve the following problems:

1. BRepFeat_MakePrism does not always give correct results when the UpToFace is not planar (and sometimes it even makes a difference whether the UpToFace is convex or concave!!!)

2. BRepFeat_MakePrism with option "2" cannot generate a correct SubShape for Pocket/UpToFirst because the subtracted material is completely inside the support - so there is no added material

3. And if we define AddShape and SubShape as reflecting not the added/subtracted material but the design intent (see separate thread I started) then we also will not be able to use BRepFeat_MakePrism any more

and will fix the following bugs:
#775
#766

# Additional information

Planned for 0.14

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-03-29 18:50
wmayer please advise on the status of this ticket. Thanks!

adrianinsaval, 2022-02-11:
If I'm not mistaken neither part nor part design use BRepFeat_MakePrism anymore, is this correct @donovaly
So this ticket would be obsolete

luzpaz, 2022-02-13:
@donovaly ok.. but not sure you're addressing the underlying issue in the report: `BRepFeat_MakePrism` was broken. Maybe the author's solution to re-write is not the right one...but what is?

donovaly, 2022-02-13:
What do you mean? BRepFeat_MakePrism is used for Pad/Pocket and Part#s Extrude and works fine.

The issue was opened 10 years ago -> lot#s have changed in OCC in the meantime.

luzpaz, 2022-02-13:
OK, thanks for the clarification.

closed 2022-02-13 by luzpaz

adrianinsaval, 2022-02-13:
IIRC there are some issues with up to face pads, but I agree that rewriting BRepFeat_MakePrism is unrealistic