FreeCAD/FreeCAD #5559: Split package into smaller ones.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=857

* **Reporter:** wmayer
* **Date submitted:** 10/28/2012
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** #ready-for-migration, addon-manager, packaging

# Original report text

A meta package freecad can be offered which installs all other packages.

First mentioned: http://forum.freecadweb.org/viewtopic.php?f=10&t=3132#p24310
Formal discussion: https://forum.freecadweb.org/viewtopic.php?f=10&t=22691

# Additional information

https://wiki.debian.org/PkgSplit

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 1/25/2022

# Discussion from Mantis ticket

### Comment by wmayer 2015-04-19 15:04
The steps to split FreeCAD into smaller packages:
1. Add a new package name to the control file (e.g. freecad-fem)
2. Write a file freecad-<PACKAGE>.install (e.g. freecad-fem.install)
Add there the files which should go to this package, e.g.:
usr/lib/freecad/Mod/Fem
usr/lib/freecad/lib/Fem.so
usr/lib/freecad/lib/FemGui.so
usr/share/freecad/Mod/Fem

3. Adjust the file freecad.install to avoid that the files appear in two packages. Instead of
usr/lib/freecad/Mod
usr/lib/freecad/lib
a more concrete list of files must be set.

---

### Comment by wmayer 2015-04-19 16:03
Use "fakeroot debian/rules binary" to avoid to build from scratch.

---

### Comment by normandc 2017-03-09 02:54
Rather than using this method, maybe look at the old freecad 0.12.5284 package from the Ubuntu 12.04 repository which didn't use any .install file but did it from the rules makefile. Attaching the debian folder for future reference as Ubuntu 12.04 will soon reach EOL and it may disappear from the packages.ubuntu.com site.

---

### Comment by Kunda1 2017-05-10 12:36
Relevant: prerequisite checker https://forum.freecadweb.org/viewtopic.php?f=18&t=21576&p=173596#p173596

---

### Comment by Kunda1 2017-05-28 12:11
Opened a dedicated Formal discussion: https://forum.freecadweb.org/viewtopic.php?f=10&t=22691

adrianinsaval, 2023-04-02:
this seems like a packaging issue more than a FreeCAD issue, given how cheap storage is nowadays this effort does not seem worth it either, if I'm not mistaken all mods are less than 200mb

maxwxyz, 2024-02-25:
Would this be still relevant or should we close this issue?

wwmayer, 2024-02-25:
Originally, I wrote the first debianization scripts to package FreeCAD and the files were part of the repository. At that time we had only one big monolithic package that contained all binaries. The idea was to split some less often used modules to further packages so that users could install what they need.

Now packaging is not our business any more and it's up to the Debian maintainers to do that (or not). So, the issues has become irrelevant in the meantime.

maxwxyz, 2024-02-25:
Thanks for clarification.

closed 2024-02-25 by maxwxyz