FreeCAD/FreeCAD #5561: Fillet can not round a face (upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=938

* **Reporter:** ghia
* **Date submitted:** 1/1/2013
* **FreeCAD version:** 0.12
* **Category:** Bug
* **Status:** assigned
* **Tags:** #ready-for-migration, fillet, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

See file in addendum of #937

While the shape is 20 mm thick, you can not place 2 fillets of 10 mm 
Fillet001 which is 9 mm can not be set to 10

# Additional information

This ticket is being tracked from [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264&p=156236#p156236) forum thread.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by wmayer 2013-07-02 03:37
Could be a float vs. double precision issue. Should be checked once the double-precision branch is merged into master.

---

### Comment by tanderson69 2013-09-23 17:29
I tried this with the current assembly branch (has float to double change) and there was no change.

I also tried this with the OCC DRAWEXE and witnessed the same behavior.

also searched OCC forum. Here is one thread.
http://www.opencascade.org/org/forum/thread_21317/?forum=3

//OCC rant removed

---

### Comment by wmayer 2015-01-14 13:34
http://tracker.dev.opencascade.org/view.php?id=25478

---

### Comment by Kunda1 2020-12-31 12:02
Can someone test with OCC 7.5.0 ?

Edit: Please remember to update the relationship with Feature #1192 if this ticket is resolved.

luzpaz, 2022-05-31:
Upstream tickets needs to be bumped and ideally added to OCC roadmap

maxwxyz, 2024-03-02:
This also applies to the chamfer tool.

maxwxyz, 2024-08-30:
Is anybody working on this issue? @wwmayer it was automatically assigned to you during the bug migration, shall I unassign you from this?

adrianinsaval, 2024-10-07:
it might be worth contacting opencascade about pricing to get this feature implemented, then we do a fundraiser to pay for it

SayakMukhopadhyay, 2026-09-12:
Seems like a fix was merged to the IR branch https://github.com/Open-Cascade-SAS/OCCT/pull/1293

stefano2734, 2026-09-12:
> Seems like a fix was merged to the IR branch [Open-Cascade-SAS/OCCT#1293](https://github.com/Open-Cascade-SAS/OCCT/pull/1293)

That is in milestone OCCT  8.1 solved issue.  
This version will come after 8.0.2. so perhaps to late to next release FreeCAD 26.3.
https://github.com/Open-Cascade-SAS/OCCT/milestone/6?closed=1

maxwxyz, 2026-09-12:
No need to comment, already marked on occ board

luzpaz, 2026-10-08:
@maxwxyz this is fixed in OCC 8.1.0 but won't be in 26.3  
Therefore this issue's milestone should be bumped to 27.1 (where hopefully OCC 8.1.0 will be available)

maxwxyz, 2026-10-16:
@luzpaz will be 27.2 earliest.

Uthopik, 2026-10-16:
[image]
Ever since the earliest versions, there’s been this error that I often encounter when the fillet or chamfer has a value that’s exactly the same as the length.

I thought that by version 1.0 or so, this would be fixed, but the error still persists.

I reported this as (Error when combining two fillets or chamfers if the value is exactly the length. #33618), but it has been closed as a duplicate of this issue #5561 from 2022.

SayakMukhopadhyay, 2026-10-16:
@Uthopik This issue was long standing due to it being present in the upstream OCC library. But, since it has been recently fixed there, the fix should land in a release or 2 (or 3).