FreeCAD/FreeCAD #5563: Free Flight View (inspection Tool)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1095

* **Reporter:** nikospap
* **Date submitted:** 4/12/2013
* **FreeCAD version:** trunk
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Most of us have played First Person Shooters games.

When you are not taking part in the battle, 
you become a "spectator",
and can move around in the 3D world,
in some sort of free flight mode.

Depending on the game you might...
a. flight/view through walls, as if you were a ghost
b. have to go around objects (e.g. walls) as if you are a 3D being yourself.

Having the following scenarios in mind,
I thing option b, would work better for FreeCad.

What's the purpose of it?

To inspect what you have created.

Let's say, you've created a set of interconnecting pipes,
and want to see how the liquid would flow in them.
With this tool you could travel in the pipes as if you were the liquid yourself.

Or you might want to create a house.
With this tool you could flight around within the rooms,
pretending to be a person walking in the house.

________________

More...
I would even go as far as suggesting the use of the standard, FPS, keyboard shortcuts,
that many people are familiar with.
Meaning, of course, WSAD.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by yorik 2013-04-13 01:40
Honestly I see little point on trying to do that in FreeCAD, when it is simple to export your work to an application more suited for that kind of thing, like Blender.

---

### Comment by nikospap 2013-04-13 11:52
So what if you want to change something in your work.
Wouldn't that mean going back to Freecad,
make the change,
then export again,
then open, again, with Blender,
and then inspect again?

Depending on what you might want to create,
how many times will you have to go back and forth
between these programs?

That doesn't seem very practical.

---

### Comment by nikospap 2014-07-28 06:42
Please, check the second video on the following link, 
where, in the John Deere factory,
they have created a virtual, 3D copy of the production line.
Amongst others, they use what I refer to as a "Free Flight View"
to inspect the production process, and make the necessary changes,
before they even start the production!

http://tinyurl.com/ky7cvkf

---

### Comment by bejant 2014-12-11 14:37
This post is to provide links to similar FreeCAD developments. 

In the John Deere video nikospap referenced it looks like the user is wearing an Oculus Rift (or some similar VR device) and dev jriegel is working on support for this:
http://forum.freecadweb.org/viewtopic.php?f=9&t=7715  

Additionally, FreeCAD forum member JMG has done something similar to a free flight view and has posted in the Users Showcase.  It allows the user to define a path on which the camera travels, and then moves the camera along that path.
http://forum.freecadweb.org/viewtopic.php?f=24&t=8437

---

### Comment by DeepSOIC 2014-12-20 00:24
+1. 
I've seen the feature in Serious Editor and I used it... only!
The standard CAD navigation is very usable for not-too-concave objects. But for exploring internal cavities that might be invisible from outside, free flight view is very handy. It's extremely hard to place the camera into a deep hole in an object with standard navigation.

axaluss, 2022-06-07:
When I started using freecad, I really missed this feature! fps style navigation is the ONLY way I know how to intuitively navigate 3D spaces.  
Adding the option should not be too hard, and especially people new to CAD would love it.

axaluss, 2022-06-07:
You could also add options
- to record a flight path (can be smoothened and finetuned afterwards)
- configure different movement speeds (maybe a button to shift between walking and running)

luzpaz, 2022-06-07:
There is some different approaches to this feature floating around on the forum. I'll see if I can find them and notate this ticket.

Anprotaku, 2023-04-17:
Do anyone have any way to get this feature?
Navigating in a FPS kinda way.

NXTler, 2023-08-26:
I also miss this feature, especially for large or complicated parts.

luzpaz, 2023-09-24:
Anyone want to take a stab at this feature ?

maxwxyz, 2024-02-25:
@Rexbas as you've fixed most issues in the navigation styles, maybe that's something for you.

Rexbas, 2024-02-25:
Hmm, maybe, I will think about it.

luzpaz, 2024-10-20:
@kwahoo2 any interest into working on this feature ?

kwahoo2, 2024-10-20:
> @kwahoo2 any interest into working on this feature ?

This does seem to overlap with some work I've done in the XR workbench. It has 2 different navigation styles, demo below:

https://www.youtube.com/watch?v=Rugrs-qlXlw

1) Arch-like movement: (like Quake with "noclip" enabled)
* analog stick/trackpad of the primary (default left) controller moves viewer up/down and left/right,
* analog stick/trackpad of the secondary (default right) controller rotates viewer around center of the HMD and moves forward/backward.

2) Free movement: (like free 6 DOF game like, hmm Descent maybe?)
* analog stick/trackpad of the primary (default left) controller moves viewer forward or backward along the controller axis,
* analog stick/trackpad of the secondary (default right) controller rotates viewer around center of the controller.

---

Something similar could be done on the "flat screen" with gamepad input and keyboard as a fall-back. I have to look how the gamepad input is done in Qt6, since Qt Gamepad seems to be obsolete. SDL might be the only sane option here.

Also I am curious if there is any active open source/open hardware space mouse development and what kind of input can it provide. Maybe there can be a common interface for gamepads and FLOSS space mice.

luzpaz, 2024-10-20:
> Also I am curious if there is any active open source/open hardware space mouse development and what kind of input can it provide. Maybe there can be a common interface for gamepads and FLOSS space mice.

I know @mnesarco had something going with https://github.com/mnesarco/HyperController but it seems to be archived project as of this moment.

mnesarco, 2024-10-20:
forget HyperController, here is one better: https://github.com/ChromeBee/Hall-Effect-Sensor-CAD-Mouse-Spacemouse

kwahoo2, 2024-10-20:
Python prototype for "Arch-like" movement: https://gist.github.com/kwahoo2/f627175344fedfb968ad37e453f8a6c1

https://github.com/user-attachments/assets/257b8853-5294-45f8-acb3-c3df05bb6f80

> forget HyperController, here is one better: https://github.com/ChromeBee/Hall-Effect-Sensor-CAD-Mouse-Spacemouse

Interesting.

kwahoo2, 2024-12-28:
I (finally) started working on this. KB+M only, implementation similar to "View turntable": small window on top. This seems to be required if we like to catch WSAD keys, otherwise W is consumed by the workbench selector shortcut. 
Any ideas for the tool's icon?

luzpaz, 2024-12-28:
@FreeCAD/design-working-group :point_up_2:

kwahoo2, 2024-12-31:
Initial implementation (not merge ready) https://github.com/kwahoo2/FreeCAD/tree/free-flight
W, S, A, D, Space, Ctrl - linear movement
mouse or Q, E, arrow keys - angular movement
The camera should be set as a perspective one.

https://github.com/user-attachments/assets/470e9361-c0d7-43e5-ad9c-3ce03433e1aa

Should work OK on Windows and Linux/X11, but Linux/Wayland does not allow to move/lock the cursor, so rotation with mouse with be limited by desktop borders. This is even worse with XWayland, as it thinks that setPos() moved the cursor, but it didn't.  I haven't found a fix for this yet.

kwahoo2, 2025-02-17:
After some conversation with DWG, I decided to ditch the separate window, creating a small widget in the 3D view instead.

[image]

after pressing Esc it extends and unveils preferences (key bindings, movement speed).

[image]

Work in progress. [Video](https://youtu.be/ii9MLq2JDio)

furgo16, 2025-12-10:
> A major disadvantage is that it introduces an additional dependency. For that reason, I would like to distribute it as external addon.

@kwahoo2 I think this could be discussed with maintainers, perhaps they find that the benefits of having this feature trumps the addition of the PySDL3 dependency. I could imagine that could be a great feature to have for the BIM workbench.

Would you be up for submitting your demo code as a draft PR for discussion?

kwahoo2, 2025-12-14:
@furgo16 

I don't want blur the discussion in more places, there is this topic, and 1 (failed) PR created.

For now I need to know if the demo code works as intended in every OS/WM configuration:

* does the mouse lock work?
* do all keys work as intended?

There is the matrix that should be filled:

| OS| Windows |Linux X11 |Linux Wayland|macOS|other OS|
| -------------| ------------- | ------------- | ------------- | ------------- |------------- |
|Mouse Lock | OK  |OK  | OK| ?  | ?|
| Keys  | OK  |OK  | OK| ?  | ?|

After that I can work on a proper SDL (pySDL) integration. Then I'll create a PR.