FreeCAD/FreeCAD #5564: Honeycomb, internal bone structure and bamboo

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1141

* **Reporter:** pfirszt
* **Date submitted:** 5/28/2013
* **FreeCAD version:** trunk
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** #tobeclosed

# Original report text

With the rise of 3D printing (cost related to volume/weight) it would be nice to have some tools to create structures like [1] honeycomb, [2] internal bone structure and [3] bamboo like structure
[1] A "sheet" on honeycomb with parametric cell size and wall thickness could be used for boolean operation.
[2] a lot off different size bubbles, bubbles don't overlap and size of a bubble depends on the distance from wall - closer to the wall -> smaller bubble. Something like a sponge, but with a "twist".
[3] in a pipe: thin internal walls with smooth translation into the main pipe wall - http://www.midwestbamboo.com/images/node_removal.JPG

1 and 3 can be made using existing tools, but 2 would require a lot of work to get a good result.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Jriegel 2013-06-26 13:01
Yes would be a nice feature, But I have no time for it....

---

### Comment by bejant 2014-02-06 03:07
I think reasonable approximations could be done with modeling some types of 3D tessellations of the M.C. Escher kind: http://www.mcescher.com/gallery/symmetry/

For case (2)above a model of a solid Box with spherical voids inside could be modeled.  The trick would be to locate the voids intersected by an edge of the box in a like manner such that when the boxes are arrayed face to face those hollow spheres align with the hollow sphere of the arrayed box next to it (for each of six faces).  The location of the hollow spheres that are contained wholly inside the box does not need to be precise.  These boxes would be arrayed to fully encase one's model, then the boolean intersection tool could be applied to remove that portion of the 3D array outside of the model.  The thickness of the model shell would fill in those voids common to the shell, and the model interior would contain the "bubbles".  The user could scale the cube model to suit his / her needs, so maybe a handful of different models would be needed, each having different solid to void ratios (densities).

Similarly, this could be done for cases (1) and (3) above but (3) would probably be more like smooth straws instead of bamboo with nodes, plates, and internodes:
http://halfsidebamboo.info/http:/halfsidebamboo.info/wordpress/96/bamboo-plant-parts-i/.

So couldn't this be achieved by some type of model library rather than coding?

---

### Comment by mrlukeparry 2014-02-08 00:17
we deal with lattice structures in our Research Group - although more to do with optimisation. I am not exactly sure how they are exactly handled as I am sure we use custom algorithms which use unit cells that form part of the tessellation. 

I am not confident boundary-surface representation is the best solution with these especially exporting to STL with tolerances and also numerous problems with minimum wall thickness from what I've heard and is often machine specific.

It'd be interesting anyway if someone on was interested in tackling this.

---

### Comment by Kunda1 2017-05-22 14:05
Honeycomb - https://forum.freecadweb.org/viewtopic.php?t=11008

---

### Comment by herrdeh 2018-03-06 13:20
I think, this gets more and more important with 3D-printers getting so cheap that even I could afford one...

I had a little success by abusing cura's support lattice feature and setting the "distance-to-walls" parameters to 0. And you can configure quite a lot in cura. The cura-support-lattice-abuse has some disadvantages:

* you cannot create holes in your lattice and empty space intended to be left empty is likely to be stuffed with lattice
* getting out the support lattice can be difficult and will leave more remainings
* you cannot create different lattice structures for support (when printing) and enforcement of your part. But you want to have as little as necessary for support - and likely much more for enforcement.

So, this feature should be implemented in FreeCAD. 
Couldn't we borrow some code (or concepts at least) from cura? - The support lattice feature seems to be quite elaborate there...

[Here](https://forum.freecadweb.org/viewtopic.php?f=13&t=27236&p=218141#p218122) was some discussion in German on that issue.

Attached photos of abused cura support lattice parts. Not pretty, but amazingly firm.

---

### Comment by Kunda1 2018-06-03 15:26
Also relevant: 
FCHoneycombMaker macro: https://forum.freecadweb.org/viewtopic.php?f=22&t=29073

---

### Comment by pfirszt 2018-12-02 15:50
Relevant python script for FreeCAD : https://reprapltd.com/fibre/

---

### Comment by Kunda1 2018-12-03 12:08
Not sure where this should go? Probably in the FreeCAD-macros repo..right ?

---

### Comment by pfirszt 2018-12-03 12:13
I guess it's a good idea.

---

### Comment by Kunda1 2018-12-05 13:25
Mentioned in https://github.com/FreeCAD/FreeCAD-macros/issues/6

adrianinsaval, 2023-04-02:
this seems out of scope here, I think this type of things are nowadays usually done by the slicer automatically, modelling the internal honeycomb in FreeCAD seems unnecessary and would likely not be as good as what slicers provide.

chennes, 2023-04-02:
I'm going to close this as "not going to implement" -- slicers do a very (very!) good job at this, and I agree with @adrianinsaval that it's out of scope for the core of FreeCAD. That's not to say an Addon couldn't implement it if someone was really keen on it, but I can't see it going into the main development.

closed 2023-04-02 by chennes