FreeCAD/FreeCAD #5565: JT File Format

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1156

* **Reporter:** templatearray
* **Date submitted:** 6/22/2013
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** feedback
* **Tags:** file format

# Original report text

Specification of the JT file format 8.1

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Jriegel 2013-06-26 09:08
There is already a JT import module in FreeCAD. but it need the JtOpenLib to work which is proprietary....

I don't think jt is implementable in OpenSource...

---

### Comment by Jriegel 2013-06-26 09:12
Anyone think we should do that? 

I think we should not....

---

### Comment by lukasubo 2016-06-15 03:27
Why does it matter so much, anyway? Is JT commonly used? Supporting interchange of proprietary file formats from a company which has a stake in their use (Siemens) is generally a bad practice, and if you have a bunch of files in this format, you can just use some other CAD (OnShape, for instance - no download, free account, and runs on any OS) to convert them to STEP.

EDIT: GrabCAD should also work as an online service to convert from proprietary file types.

---

### Comment by lukasubo 2016-06-15 03:29
Oh, I just realized I bumped this from 3 years ago... sorry. I saw it in the "Unassigned" list and didn't check the date.

---

### Comment by Kunda1 2017-03-02 12:07
yorik do you mind weighing in on this?

---

### Comment by yorik 2017-03-02 13:16
It is very unlikely to be implemented, but we can also just leave the ticket open, someone might get interested in working on this in the future...

adrianinsaval, 2023-04-02:
10 years without any prospect of being implemented, only meaningful comments talks about how this is not worth doing... I suggest closing this.

chennes, 2023-04-02:
Agreed.

closed 2023-04-02 by chennes