FreeCAD/FreeCAD #5567: Adding an option to fill a the surface with image or other methods to show a thread-like shape

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1181

* **Reporter:** project4
* **Date submitted:** 7/14/2013
* **FreeCAD version:** trunk
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

I saw in other applications that it's common to fill the surface with an image that looks like a thread on parts.
That simplifies the object since there will be much less details than modeling the threads on a bolt or nut.

Other ideas to use an image on a part is to apply a material on it.
I think I saw somewhere that material is one of the big features for FreeCad, so I'm sure there will be much more work done in that direction, but the threading appearance is not related to that, so I hope you could add at least that property to the part designer.

Thanks.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/20/2021

# Discussion from Mantis ticket

### Comment by wmayer 2013-07-16 12:01
Do you talk about texture mapping?

---

### Comment by project4 2013-07-16 15:19
Yes, I think that's how it's called.
But I'm talking about one specific case for the threading and not every possible image to be used as a texture.

---

### Comment by shoogen 2013-07-17 03:00
This could be useful for knurling as well.

---

### Comment by ulrich1a 2013-07-30 06:04
There is a need to mark holes or bolts, which have a thread, in some way. Because if we want to have a drawing of a part with a thread, we also want the thread shown on the drawing with its characteristic drawing style.
This method is named "cosmetic thread" in solidworks. It is shown for example at this page: http://help.solidworks.com/2013/English/SolidWorks/sldworks/c_cosmetic_threads.htm 
I think there should also a way to have this in a step file, but I could not find any useful information.

---

### Comment by Kunda1 2017-01-19 15:27
Changing status back to new since OP responded. 

---

### Comment by Kunda1 2021-02-20 22:45
davidosterberg how much does this ticket related to Helix ?

PLChris, 2022-05-12:
New discussion respecting the developments since then, see https://forum.freecadweb.org/viewtopic.php?f=8&t=68646

luzpaz, 2023-11-07:
@PLChris do you mind updating this ticket with the more up-to-date issues that are wanted ?

pierreporte, 2023-11-10:
This issue should be called: “Add support for cosmetic threads”. This is an essential feature but, more importantly, it should be linked with the drawing. Also, a cosmetic thread should accept to have material removed afterwards and that should be shown in the drawing as well. This is essential for reliefs and mentioned in the forum thread.

Below is a comparison between Solid Edge and Alibre. They handle cosmetic threads a little bit differently but none of them get it perfect every time. Solid Edge is the best but Alibre has some nice perks while having terrible issues.

The second picture is from the forum and shows a good representation. It is good because it shows the thread pitch and has a color matching with the part. Other software (Solid Edge, Alibre, Solidworks) don’t change the color. This is very minor.

The cosmetic threads should have an option to be shown either like this (a nice representation with fake threads) or ever more simply to save computing time. It could be activated by users when they have a lot of bolts, nuts and other threaded parts. It would be acceptable to have only the simplistic representation instead of the texture.

The 3 most important points are:

* users must be able to distinguish the threaded zones in the 3D view;
* the cosmetic threads must be automatically inserted in the drawings;
* materials removed from threads in 3D must not show anything in the drawing.

Of course everything should work for external and internal threads.

[image]

[image]

pierreporte, 2023-11-13:
Issue #5560 may be a prerequisite for some possible implementations of this feature.

luzpaz, 2023-11-13:
Thanks for the assist @pierreporte

pierreporte, 2024-03-05:
@martinproks is trying to solve this, first as a feature in the tree and 3D view. He acknowledges the need for the translation of cosmetic threads into drawings.

Repository: https://github.com/martinproks/cosmeticthread3d

Forum thread: https://forum.freecad.org/viewtopic.php?t=85047

pierreporte, 2024-03-05:
On top of cosmetic threads, there could be cosmetic knurling as well. Similarly, the precise representation need intensive computation and they appear in a simpler form in the drawings. There are several standards: DIN 82 and ISO 13444 in the metric world.

The most used are the axial and crossed 30° types. The depth can vary a lot but this should be left for the drawing.

In niche industries like horology, not only cylinders and planar surfaces are knurled: toroidal (convex and concave) and conical surfaces can be knurled as well. Knurling rollers are sold in a wide variety of shapes.

[image]

MiqCG, 2024-06-23:
If it helps, here is another macro that makes cosmetic threads:
https://github.com/andesfreedesign/CosmeticThreads_FreeCAD

Author @andesfreedesign

martinproks, 2024-06-30:
Hi, I know this and I tried it.

I had a little pause from FreeCAD (not only FreeCAD is life about ;-) ), but now I'm back to move it forward...

maxwxyz, 2024-09-04:
@martinproks any news on this?

maxwxyz, 2024-10-05:
@pierreporte IMO this makes sense with external threads implemented (should work the same as holes for displaying) https://github.com/FreeCAD/FreeCAD/issues/12679

pierreporte, 2024-10-05:
Yes, the feature is necessary for both internal and external threads.

pierreporte, 2024-10-14:
@pieterhijma Why did you remove the status in the board?

pieterhijma, 2024-10-15:
@pierreporte, My apologies. I'm not aware that I changed anything. I did browse the board and studied the priorities and tried to filter. I remember that I clicked on this particular status to see what different options there were. I'll restore it to Todo (P1) which it was before.

pierreporte, 2024-11-29:
Here is a UI/UX proposal: https://design.penpot.app/#/view/a0a8e792-b2d2-818e-8005-594ea94fab2e?page-id=a0a8e792-b2d2-818e-8005-594ea94fab2f&section=interactions&index=0&share-id=7ad540b5-8190-815d-8005-595da691d1ca

It includes a new hole tool.

pierreporte, 2024-12-03:
@luzpaz No solution have been proposed, only a UI proposal.

I will write here more details about the desired solution.

pierreporte, 2024-12-14:
As discussed in Developers meeting on 1st December, here is a writing describing the desired feature.

--------------------------

# Cosmetic threads

## Current situation

In the Part Design workbench, the Hole feature is able to produce threads. These threads are modeled in 3D, that are resource-heavy (the generation of a single threads takes a short but noticeable amount of time). When the checkbox to model the threads is unchecked, the representation is simply the drilled hole, with no visual indication that it the hole is threaded. There is no feature to create external threads, or to add internal threads to existing holes (although the Hole feature can be used as a workaround). The Part workbench doesn’t provide threading features.

In the TechDraw workbench, threads created in the Part Design worbench are not recognized as such, whether they are modeled or not. Modeled threads are represented as they are visible in the 3D view. The workbench provides features to add standard threads symbols, both in top view and in side view, to dress up cylindrical features but it is not an automatic process and it doesn’t produce accurate results (for example, the diameter of a M3 thread annotation isn’t exactly 3 mm in diameter). The nature of these annotations is not taken into account by dimensioning tool.

## Feature set

FreeCAD needs threading feature that can be applied in Part Design and Part workbenches without the need to model them while providing a visual cue of the existence of threads (hence the word “cosmetic”). Besides the integration of threads into the Hole feature, there should be standalone features to create external and internal threads on existing geometry. These new cosmetic threads must be understood by TechDraw that would generate automatically the correct annotations, compliant to drafting standards. Dimensions referencing threads annotation must be correct and display the right designation.

When material is removed from a cosmetic thread (for example an undercut near a shoulder), the removed section of the thread must be removed from the representation in TechDraw. It may seem obvious, but while most CAD application do understand correctly removal of material, Alibre doesn’t and is thus hardly usable in this regard. It is important that FreeCAD doesn’t fall in the same trap.

The Part Design and Part feature must allow the user to create both cosmetic threads and modeled threads. It should be possible to switch all threads of a part from cosmetic to modeled (and back) using a simple action (like an entry in the context menu of the body).

The Render workbench could make use of the feature by using a special texture for the threads to generate realistic appearance, or ask the user to switch to modeled threads for higher realism.

The features should be compatible with directional threads (e.g. buttress threads) or allow them to be easily implemented in a later stage.

The cosmetic threads should be understood by the CAM workbench to select the right tool automatically.

The features could be generic enough to allow an implementation of knurling using the same mechanism (see ISO 13444).

## Standard threads

This section lists common thread standards and gives relevant input parameters. They can be sorted in the following categories:

- General use threads: metric, unified inch
- Aerospace threads: aerospace metric, aerospace unified inch
- Piping threads: BSP parallel, BSP tapered, NPT, NPSC, NPTR, NPSM, NPSL
- Trapezoidal threads: trapezoidal metric, ACME

### Metric (M)

Thread shape per ISO 68-1.

Parameters:

- Size (diameter): choice from list per ISO 261 (default) or numeric input
- Pitch series (standard size only): coarse (default), fine, or custom
- Pitch (non-coarse only): choice from list (default) or custom numeric input
- Direction: right-hand (default) or left-hand
- Tolerance class (optional): choice from list per ISO 965-1 (default), or custom numeric input

When pitch is selected from a list, the largest value is the default

Examples of designations: M3, M8×0.5 LH, M10-6g, M100×1

### Unified inch (UN, UNR)

Thread shape per ASME/ANSI B1.1.

Parameters:

- Size (diameter): choice from list per ASME/ANSI B1.1 (default) or numeric input
- R-type: no (default) or yes
- Pitch series (standard size only): coarse (UNC, UNRC) (default), fine (UNF, UNRF), extra fine (UNEF, UNREF), or custom
- Threads per inch (custom only): numeric input
- Direction: right-hand (default) or left-hand
- Tolerance class (optional): choice from list per ASME/ANSI B1.1 (default), or custom numeric input

When threads per inch is selected from a list, the largest value is the default.

It is not necessary to name pitch series in the list (UNC, etc.) and they are indicated for reference only. “Coarse”, “Fine”, and “Extra fine” are enough.

[... 149 lines cut ...]
## UI / UX

The standalone thread feature has a different interface than the integrated thread capability of the Hole tol.

### Standalone thread feature

Catia was the most important source of inspiration. [Its documentation](http://catiadoc.free.fr/online/prtug_C2/prtugbt0616.htm) show how it works. The screenshot below comes from there.

[image]

The first step is to chose the location of the thread. The user first selects the lateral face of the thread (either a cylinder or a cone). The feature automatically detects if the surface is concave or convex to determine if the thread will be external or internal. Then, the starting p [... cut ...]
The firsrt selection is the sketch containing all the centers for the new hole pattern. Then the user selects the hole type (straight hole, countersink hole, etc.). At this point, the geometry of holes can be defined.

The thread is an option, that, when it is not activaed, is completely folded. Unlike, the current Hole tool, it is not fully visible unless activated to keep the task panel as clean as possible. When activated, the drill diameter is set automatically for the selected thread, unless the user wants to chose a custom value.

pierreporte, 2025-01-07:
In Solidworks, there is a dashed lines displayed around threads to show the thickness of threads. This could be useful to help users estimate how much material is left when they add or remove material near threads.

[image]

Also in Solidworks, recent versions introduced a normal map texture to increase realism of cosmetic threads.

[image]

alfrix, 2025-01-14:
did a test of the mockup based on #19052

i've drawn the svgs myself, and there is a lot of cleanup and tweaking still, but it looks promising after only a morning

[Videocaptura de pantalla_20250114_151138.webm](https://github.com/user-attachments/assets/1c615d54-c067-4ca5-8412-62b33936860b)

pierreporte, 2025-01-14:
@alfrix That’s awesome! Could you please post this as a comment in #19052 so we can discuss there? Issue #5567 is only for cosmetic threads.

pierreporte, 2025-08-17:
The specification written in https://github.com/FreeCAD/FreeCAD/issues/5567#issuecomment-2543142577, including dedicated threading features, automatic TechDraw support, new thread system, and UI/UX overhaul will be transformed into a dedicated [FEP](https://github.com/FreeCAD/FreeCAD-Enhancement-Proposals).

pierreporte, 2025-08-21:
The standalone thread feature has been moved to a separate issue: https://github.com/FreeCAD/FreeCAD/issues/23257

maxwxyz, 2026-02-07:
From the 2026 complaint session:
**We need a way to get a listing of options that are not set to their defaults**

pierreporte, 2026-02-07:
I don’t understand what the complaint has to do with cosmetic threads.