FreeCAD/FreeCAD #5568: Option for PolarPattern to obey individual UpToFirst/UpToLast/UpToFace pad features for each instance.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1190

* **Reporter:** vnks
* **Date submitted:** 7/27/2013
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** PolarPattern

# Original report text

Hi, 

Applying a polar pattern to a pad feature in Part Design, it appears that the length is taken from the original feature instead of padding each individual rotated feature to first/last/specific face. In the attached design, if you look inside the hollowed out part, you can see that the two poles, meant to hold screws, are the same height, instead of extending up to the 'ceiling' per 'UpToFirst' pad feature.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-05-31 22:19
Abdullah when you get a moment please check this ticket. Thanks!

---

### Comment by abdullah 2017-06-10 23:29
kunda1

I think I understand what the reporter wanted to do. I am not sure how this should actually work. 

The first pad goes up to first. Now, I am not sure whether the transformation should "recompute" where it should end or not. I mean, I understand it would be useful for the case of the reporter. I simply do not know how general such a case is.

I think it is worth to involve one of the CAD experts to gather more information about user requirements.

---

### Comment by normandc 2017-06-17 21:46
I'll test in the commercial CAD software I use at work and report back.

---

### Comment by normandc 2017-06-17 21:49
FYI, there is a simple workaround here since there are only two occurrences, delete the polar pattern and add a second circle in Sketch016031002. Two posts will be created up to the face of the inner loft.

---

### Comment by Kunda1 2017-06-21 10:51
vnks please respond

---

### Comment by normandc 2017-08-02 11:02
Forgot to report back. In the commercial parametric CAD program I use at work, the behaviour is as the reporter is asking. The transformation is recomputed according to the ending condition of the original pad.

Maybe this could be considered an enhancement request rather than a bug.

---

### Comment by Kunda1 2017-08-02 14:00
Changed to 'Feature'

---

### Comment by Kunda1 2018-05-31 00:31
Opened forum thread for further discussion: https://forum.freecadweb.org/viewtopic.php?f=8&t=29035

---

### Comment by Kunda1 2018-10-08 14:17
NormandC how would you rename this ticket to reflect that it's a feature instead of something missing in FC ?

---

### Comment by Kunda1 2019-03-03 14:19
Renamed ticket title per the chrisb's recommendation in the forum

FlachyJoe, 2023-11-27:
@luzpaz I agree something has to be done but it needs a logic change in PartDesign::Transformed which currently acts on shapes not features.

maxwxyz, 2024-02-25:
Is this related to https://github.com/FreeCAD/FreeCAD/issues/11135 or https://github.com/FreeCAD/FreeCAD/issues/11111 ?

FlachyJoe, 2024-02-25:
@maxwxyz the two issues you mention can be solved without the huge rewrite this one requires.

FlachyJoe, 2024-12-07:
Still an issue