FreeCAD/FreeCAD #5569: Full round fillet

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1192

* **Reporter:** din743
* **Date submitted:** 7/31/2013
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

Option to create a fillet that is tangent to three adjacent faces.

# Additional information

Example from another CAD package: http://help.solidworks.com/2013/English/SolidWorks/sldworks/r_Full_Round_Fillets.htm

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by normandc 2013-08-02 19:13
This might require to fix bug #0000938 first.

adrianinsaval, 2023-03-22:
this is basically not doable unless occt implements it and has been inactive since 2013, I suggest closing

grubertechnic, 2023-07-08:
@adrianinsaval : at least it is a duplicate of #5561. Not 100%, but same origin, no full fillet possibilities in OCCT. So i also suggest closing.

adrianinsaval, 2023-08-07:
Is not really the same as that one, but since it's out of scope and not a useful ticket I'm closing anyways.

closed 2023-08-07 by adrianinsaval