FreeCAD/FreeCAD #5570: Parts Appear as Shells Instead of Solids When Viewed by Clipping Plane

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1208

* **Reporter:** bejant
* **Date submitted:** 8/10/2013
* **FreeCAD version:** 0.13
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

FreeCAD version is actually 0.14 (0.14 presently is not selectable from Product Version box).

I do not know if the parts are actually solids that appear as shells, of if the parts really are hollow shells.  Link to forum thread:

http://forum.freecadweb.org/viewtopic.php?f=3&t=4371

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 5/9/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-01-13 13:29
Some developments with coin3d has made this a possiblity: Please see http://forum.freecadweb.org/viewtopic.php?f=3&t=4371&start=10#p94838

---

### Comment by berndhahnebach 2019-03-01 21:32
related https://forum.freecadweb.org/viewtopic.php?f=8&t=18188

---

### Comment by dstarke 2021-05-09 20:42
Create a rectangular filled face in the Draft workbench and use that with the "Slice apart" tool of the Part workbench also results in a solid clipping plane.
See https://www.xsim.info/articles/FreeCAD/en-US/HowTo/Cut-a-solid-with-a-plane.html

adrianinsaval, 2022-06-21:
@realthunder if i'm not mistaken this issue is solved in your branch, do you plan on doing a PR for that this cycle? or does it require your experimental renderer?

realthunder, 2022-06-22:
Yes, it requires the experimental render caching, which wold be a big PR of 10K+ lines of code.

Roy-043, 2023-07-16:
Note that since V0.20 we have the [Part_SectionCut](https://wiki.freecad.org/Part_SectionCut) command.

NXTler, 2023-08-26:
Couldn't we put the Part_SectionCut feature into the sketcher for example?

maxwxyz, 2023-09-26:
The Part_SectionCut is a heavy operation and takes long to load on complex geometry. I would prefer a switch in the clipping plane or settings panel and a default setting for a solid looking behavior (not a shell) in the clipping plane and sketcher.

maxwxyz, 2023-10-24:
As this issue comes up frequently, is there any other fix than to implement the "big PR" or is there a timeline when the experimental render caching could be merged @realthunder ?

@adrianinsaval  I am not a programmer but found https://dev.opencascade.org/doc/overview/html/occt_user_guides__visualization.html#occt_visu_4_4_16

There the following is mentioned:
- set capping and capping material
- capping texture (hatching)

https://dev.opencascade.org/doc/refman/html/class_graphic3d___clip_plane.html
Is this a "simple" fix or more difficult to implement?

adrianinsaval, 2023-10-24:
we don't use occt's rendering engine so I don't think anything mentioned there helps us.

One potential workaround is to generate an intersection between a face coincident with the clipping plane and the displayed geometry. This a less intensive operation than making a full boolean cut but it can still be demanding. I tested the difference in a model with 800 extruded faces using recompute profiler macro:
```
18:03:02  1 features are touched
18:03:02  Generating execution order...
18:03:02  Number of features to execute: 1
18:03:02  Execution order:
18:03:02      Cut
18:03:02  Recomputing... (time in seconds, label)
18:03:02  	43.783663272857666	Cut
18:04:19  1 features are touched
18:04:19  Generating execution order...
18:04:19  Number of features to execute: 1
18:04:19  Execution order:
18:04:19      Common
18:04:19  Recomputing... (time in seconds, label)
18:04:19  	11.138192415237427	Common
```
it will likely still generate lag when defining the clipping plane but it might be tolerable. If implemented it should definitively have a way to disable this

maxwxyz, 2023-10-25:
I've come across a Python / OpenGL implementation: https://forum.freecad.org/viewtopic.php?style=10&t=18188&start=10
Is this feasibly to integrate?

yorikvanhavre, 2023-10-27:
It is feasible, but there is apparently still a lot of work needed to be functional and reasonably bug-free...

maxwxyz, 2024-02-25:
@davesrocketshop sorry for asking but would there be a hack to use some material properties from your overhaul to make the section cut look like a closed profile?

davesrocketshop, 2024-04-15:
> @davesrocketshop sorry for asking but would there be a hack to use some material properties from your overhaul to make the section cut look like a closed profile?

Not really. It'll need to be addressed properly first

adrianinsaval, 2024-04-15:
I get the impression that simvoleon is rather aimed at rendering somthing like a FEM mesh to visualize de value of a field inside the volume, we want something much simpler which is just a plain 2d section of a volume

yorikvanhavre, 2024-04-16:
Yes I don't remember why I thought otherwise, but looked at the simvoleon code now and indeed there is nothing aside volumetric data. Probably nothing we can use there.

DrD4ffy, 2024-05-07:
When using clipping plane the model looks hollow because the face that closes the cut is missing. A manual workflow using Part_CrossSections can provide this missing face:

0.) Original (2 bodies)
[image]

1.) Part WB -> Cross-Sections
[image]

2.) Part WB -> Make face from wires
[image]

3.) Draft WB -> Hatch
[image]

4.) View -> Clipping plane
[image]

So FreeCAD seems to already have the needed functionality to create this missing geometry. Couldn't this be integrated in the Clipping Plane function?

NXTler, 2024-05-07:
Now that I see it, having a hatch for the clipping plane seems really nice and easy to understand. When we have this feature some day, I would like to have the option for this.

yorikvanhavre, 2024-05-07:
> So FreeCAD seems to already have the needed functionality to create this missing geometry. Couldn't this be integrated in the Clipping Plane function?

The problem is that this relies on a heavyweight OCCT calculation, while the current clipping tool works at coin level and is therefore super fast, which is what we need for a "visual" tool.

I think the best solution would be something sketched in https://forum.freecadweb.org/viewtopic.php?f=8&t=18188 : extend the coin clipping plane code with openGL code to render the clipped area. The problem is, who has the skills to do that :sweat_smile:

davesrocketshop, 2024-05-07:
> The problem is that this relies on a heavyweight OCCT calculation, while the current clipping tool works at coin level and is therefore super fast, which is what we need for a "visual" tool.
>
> I [...]

That is pretty much identical to the work for textures. Maybe I was too quick to dismiss it earlier, but it could be revisited post 1.0 when I get back to per face texturing

GAZ082, 2024-11-23:
This would be so nice for 1.1

maxwxyz, 2024-11-23:
@shaise I'm assigning you as you've made some demos on the forum. Feel free to remove your assignment if you are not working on this.

shaise, 2024-11-23:
Great!
I just need time to get into it. I'm currently working hard on sheetmetal to make it 1.0 worthy.

maxwxyz, 2025-04-16:
@shaise any news so far?
what I want to add: it would be ideal if at the plane where the parts are clipped, an outline should be visible (currently if you cut a face, there is no outline). Ideally you could select, reference and work with that outline (future implementations)

[image]

pierreporte, 2025-04-16:
@maxwxyz IIRC there is a separate issue about the outline. It is definitely needed.

pierreporte, 2025-04-16:
@GAZ082 I checked and it appears he didn't open a grant request. There is no proposed reward for this either, as they were targeted for other things.

GAZ082, 2025-04-16:
@pierreporte @maxwxyz  dont know how the reward proposal works, but from what I read from Max in this issue, there have been several other requests to improve the coin based clipping pane. I think this is a good topic to offer a bounty. Heck, i'd even chime in some money for this :)

GAZ082, 2025-04-16:
@shaise 
https://fpa.freecad.org/handbook/process/grant-program-rules.html

If you need an incentive to put this at the top of your queue, looks like that if the amount is reasonable it may be approved.

shaise, 2025-04-18:
Hi everyone. 
1.There are some works in progress (some of the grants) that address coin and rendering system. I want to wait until finished.
2. No need for grants. All my work for freecad is completely voluntary. And will stay so. 
3. Though I made some previews for the cross section fill, l'm not completely familiar with freecad internals. I know how to setup the shaders but l need a way to get a list of all cross sections and their location. It will be great if someone can help me with that.
Shai

maxwxyz, 2025-04-18:
@kadet1090 @davesrocketshop maybe regarding shaders?
@pieterhijma maybe there is also something from OCC?

pieterhijma, 2025-04-21:
> [@pieterhijma](https://github.com/pieterhijma) maybe there is also something from OCC?

After reading through the forum post and the comments, I don't think this is an OCC issue, but more rendering.

kadet1090, 2025-04-21:
This is related to coin and requires custom shaders / opengl code to work.

cNori, 2026-06-17:
@shaise well i have given up on trying to get it to the source integration,  i did have deep look at the renderer code is spaggery actually one of the worst i seen can't do clean implementation in any shape or form to much mixed code (logic rendering code + export in same file) and im geting lost + very old open GL API use dose not help ,
i wish good luck if some one likes to try it , i dont have time to spend 1 month do go deep in to this and re write a bunch of stuff or try to add to spaggery

GAZ082, 2026-08-27:
>  i dont have time to spend 1 month do go deep in to this and re write a bunch of stuff or try to add to spaggery

Only 1 month? Apply for a grant and get it done! :)

ahs3n, 2026-08-29:
I was thinking it would be as simple as adjusting the fragment shader of the mesh to return a solid colour or screen space pattern if the triangle is backfacing. Line rendering would also need to be disabled for backfacing faces. From the discussion here I'm not really sure what's going on 💀

I'm guessing the line rendering runs after the shading pass, using its depth buffer, because in clipping view, the clipping edges flicker between having lines and not. If that really is the case, it could be as simple as overwriting the depth buffer with the clipping plane depth if backfacing. This might just prevent line rendering automatically and would be simple enough. For multiple planes use the max depth of both planes. If it is that easy to overwrite the depth buffer, then it should be easy enough to set the normal to that of the plane as well and allow the shading to continue as normal, or shade with a world space hash pattern.

Another method would be to do the same things but without writing to the depth buffer. The hash pattern option would then have to be in screen space, and line rendering disabled for backfacing areas some other way.

For the case of a view ray entering from one plane and exiting from another, I dunno. Generating the clipping face as mentioned by others would eliminate with this specific case, at the cost of performance. Clipping view is essentially free as part of the typical rendering and it would be a shame to lose that.

Pretty simple in theory, no idea how difficult it would be in the FreeCAD codebase.

shaise, 2026-08-30:
@ahs3n , this was exactly my idea as I tested it here: https://www.youtube.com/watch?v=YGmcQ8SxeOU
It works fine, no problems with line rendering, however integrating it correctly with freecad's core code was a bit over my league.

ahs3n, 2026-08-30:
I see. graphics spaghetti code can be quite convoluted

GAZ082, 2026-10-02:
Someone please write a grant proposal and the money needed. There is thousands of euros available.

GAZ082, 2026-10-02:
Also, this ticket most likely wont be ready for 26.3 @maxwxyz

greggjaskiewicz, 2026-10-02:
If anyone wanted to pay me to work on https://github.com/FreeCAD/FreeCAD/pull/28647 - it probably would have helped taking time off day-work.

franzbu-png, 2026-10-11:
On top of this, I added an additional visualization to the already existing clipping plane here: PR #33186 , which could be used as an addition to @greggjaskiewicz's #28647  , instead of creating a section geometry, this will only affect the rendering. 
Let me know whether this is interesting to you

[image]