FreeCAD/FreeCAD #5571: Welding WB (which also includes Technical drawing symbols)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1240

* **Reporter:** nikospap
* **Date submitted:** 9/11/2013
* **FreeCAD version:** 0.12
* **Category:** Feature
* **Status:** new
* **Tags:** #ready-for-migration

# Original report text

Think of iron welding, hot glue, silicon sealer...

Because the welding material is an object of its own, and its volume is not insignificant.

Like "fillet" or "chamfer", but where two elements meet.

And by elements I mean, a point, a line, a surface.

We would select two of the above and we would apply "weld".

We would determine the thickness of the weld material,
and the allowed, maximum, possible gap between them.
Because these two elements might cross each other,
or there might be a little space between them (for example 1mm).

The weld would be applied on the "free" areas.

It must be graphically clear that this is a weld.

Useful for both the design and the construction of a project.

# Additional information

I 've made two pictures, showing how weld might look like.

And there is a third picture.
I can't imagine how Freecad would calculate the welding, for this case.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/3/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2017-07-19 11:06
yorik perhaps you still have these attachments somewhere (fingers crossed) ?

---

### Comment by Kunda1 2017-07-19 11:16
nikospap perhaps you could find them? ~9782

---

### Comment by nikospap 2017-07-21 08:11
re-uploading attachments

---

### Comment by nikospap 2017-07-21 08:42
After a long time, I found out that there are actually "weld symbols" used in drawings.
See...
https://yandex.com/images/search?text=weld%20symbols

I attach a properties dialog from WeldCAD
(a program that is no longer supported, but still available in http://arcworks.lincolnsupport.com/files/wcad3b_inst.exe , according to http://www.weldingdata.com/weldcad/weldcad.html ).
Just to get an idea.

Of course this is not the same as my initial request. I think that these symbols are "official" and more advanced, used in professional drawings. My suggestion seems to be more suitable for amateur users.

---

### Comment by Kunda1 2017-07-21 09:25
Appropriate forum threads: 

*  [Welder looking to be more efficient using FreeCAD](https://forum.freecadweb.org/viewtopic.php?f=3&t=21374)
*  [The Big TechDraw Comments Thread](https://forum.freecadweb.org/viewtopic.php?f=35&t=16831&p=165213&hilit=welding#p165213) mentioning Welding symbold from Drawing Dimensions WB
*  [I've lost the Welding Symbols of the Drawing Dimensioning Workbench](https://forum.freecadweb.org/viewtopic.php?f=3&t=19533&p=151181)
*  [Gui Commands with the QActionGroup widget](https://forum.freecadweb.org/viewtopic.php?f=22&t=9822&p=79590) - hamish talks about adding welding symbols

---

### Comment by normandc 2017-08-04 21:51
Adding welding symbols to the TechDraw workbench (TechDraw will replace the Drawing workbench which is no longer developed) is a matter for a different but linked report. Linked because the weld symbols would need to extract data from the actual weld objects on the 3D model.

---

### Comment by Kunda1 2017-08-05 13:08
Thanks normandc for your comment. Do you mind opening and linking said report?

---

### Comment by Kunda1 2018-03-03 18:15
Related: '[Simulated Weld](https://forum.freecadweb.org/viewtopic.php?f=24&t=27240)' in the FC User Showcase subforum

---

### Comment by Kunda1 2018-11-16 02:34
wandererfan should I move this to Techdraw based on what NormandC wrote in ~9873 ? (and rename the issue?)

---

### Comment by wandererfan 2018-11-16 14:45
Kunda1 symbols on drawings would be TD, but if the welds are to be modelled then that would not be TD.  

Ultimate solution probably needs both, so I would leave things as they are. 

caveat: IANAW (I Am Not a Welder)

---

### Comment by nikospap 2018-11-17 09:57
But if you had to choose between the two methods, you should choose the professional symbols. Simply because they must be more well though, and more thoroughly tested.

---

### Comment by Kunda1 2019-03-29 01:35
reox writes in https://forum.freecadweb.org/viewtopic.php?f=35&t=34990&p=297616#p294888
[quote=reox]I had a quick look around, but it looks like they are not defined in unicode.
But someone created a glyph font: https://www.fontspace.com/micronus/y145m-2009/13390/charmap[/quote]

---

### Comment by Kunda1 2019-08-14 19:04
Welding Symbol Testing Help - https://forum.freecadweb.org/viewtopic.php?f=35&t=38560

adrianinsaval, 2023-03-22:
do we want to accept request for whole new workbenches? IMO we already have too many WBs, so I would by default deny requests for new ones unless there's really lot of desire for this in the community (e.g. assembly wb)

adrianinsaval, 2023-04-14:
I'm closing this issue as it has been sitting for years without meaningful activity, I think welding symbols are available in techdraw now and the initial request to model the welds in 3d seems very very niche so in any case it would be better to make this as an addon.

In general I believe we are not looking to add more workbenches to the core unless there is someone willing to make and maintain it, with significant community support and compelling use cases.

If there are any interested people in the functionality vaguely described here you may create issues for the individual functionality (rather than a full blown workbench), describing what is missing from existing functionality and what are the use cases for such functionality

closed 2023-04-14 by adrianinsaval