FreeCAD/FreeCAD #5572: Smart tools for part design (Grow, Lattice, Snap&Lock, Split to mill, Convert to Frame, Puzzle)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1262

* **Reporter:** mntc
* **Date submitted:** 10/1/2013
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** snap, split ticket

# Original report text

This can be a new group of automatic design tools.

1) "Grow":  Sometimes up to 99% of engineers work is dedicated to bulding bulk bridge connection between active components (motors, actuators, etc). Sometimes it is boring and evident work. What if we could specify contact faces and spatial limitations, and "grow" between them part like a plant (may be with help of cellular automata mathematics). There can be preferences like "optimize weight/stiffness/volume/etc" or "style: frame/tree/box/etc"

2) "Lattice": used like "Pocket", but not cuts, but converts corresponding volume to 3D lattice of arbitrary pattern. It can easily produce springy and spongy areas in 3D-printable parts.

3) "Snap&lock": convert bulk part (with only flat faces) to set of flat walls (with used defined thickness) assembled by snap'n'lock principle. It can make easy design of lasercut boxes.

4) "Split to mill": splits freeform part to set of parts which can be milled on common 3D CNC milling machine fron one side. E.g. sphere will be split to 2 halves.

5) "Convert to Frame: reduces weight and material volume by substracting lots of material with specific algorithm

6) "Puzzle": splits big flat parts to small fragments with puzzle-like connections. Used to make big things on small CNC and lasercut machines.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-05-07 14:48
kkremitzki should I split this ticket?

---

### Comment by kkremitzki 2017-05-07 23:03
Kunda1 No, I think it's fine as is. If one of the individual tools starts getting worked on, then it would be appropriate to create a new issue to track that effort.

---

### Comment by Kunda1 2017-05-26 21:31
Per DeepSOIC 
[quote="DeepSOIC"]
I think the suggested "Lattice" feature is impractical, because it will generate a very heavy and complex b-rep, that will not help at all in the end because it will simply confuse slicers.

I think such a feature can make sense only in tight integration of some slicer, in a form of creating a compsolid-like geometry.[/quote]

adrianinsaval, 2022-03-10:
Link to forum discussion? I can't find it. Looks like a request too broad.

adrianinsaval, 2023-04-21:
dormant for 10 years and out of scope for Part Design, consider developing as an addon instead.

closed 2023-04-21 by adrianinsaval