FreeCAD/FreeCAD #5575: Configure light sources

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1294

* **Reporter:** wmayer
* **Date submitted:** 10/31/2013
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

http://catiadoc.free.fr/online/basug_C2/basugbt1701.htm

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 4/21/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-08-10 16:59
relevant: https://forum.freecadweb.org/viewtopic.php?f=10&t=9663&p=326080#p326080

---

### Comment by Kunda1 2021-04-21 18:10
realthunder has implemented this in his LinkStage3 branch, here's the initial post he made about it: https://forum.freecadweb.org/viewtopic.php?f=10&t=9663&start=60#p394930

maxwxyz, 2023-10-22:
Is there a chance to make a PR from @realthunder to FC main for this, which should fix https://github.com/FreeCAD/FreeCAD/issues/11113

closed 2023-10-22 by wwmayer

wwmayer, 2023-10-22:
According to the description the changes from realthunder is something else. This ticket is about changing the direction of the light source which by default is the negative Z axis, i.e. (0,0,-1).
Actually the possibility to modify the directional light has existed in FreeCAD for many years but there was no easy way to change it. With https://github.com/FreeCAD/FreeCAD/pull/11146 this is fixed now.