FreeCAD/FreeCAD #5576: Request for additional (dynamic) detail in error messages

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1343

* **Reporter:** ildawson
* **Date submitted:** 1/3/2014
* **FreeCAD version:** 0.13
* **Category:** Feature
* **Status:** new
* **Tags:** exception, Solver

# Original report text

When attempting to place a constraint, if the incorrect parts of the sketch are selected an error message will be displayed (for example when selecting the InternalAngle constraint the message "Wrong Selection: Select exactly one or two lines from the sketch." is displayed). It would be helpful is the message could detail what was selected (for example "One line and two points have been selected.") This would make it easier to correct/understand your mistake.

In general when an error message indicates that a user has not performed an action correctly, and details what the user is required to do, the message should also detail what inputs the user entered to better train/educate the user.

# Additional information

This feature will be more useful to beginner users as they learn how to apply constraints. 

Version Details:
OS: Windows
Platform: 32-bit
Version: 0.13.1828
Python version: 2.6.2
Qt version: 4.5.2
Coin version: 3.1.0
SoQt version: 1.4.1
OCC version: 6.3.0

# Steps to reproduce

1. Enter sketch mode
2. draw a line.
3. Select the line
5. Select InternalAngle constraint on toolbar. (works on other constrains as well)
3. Observe error message.

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **OS: Windows Eight**
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-08-28 14:33
Abdullah did any of your Error throwing logs address this ticket ?

---

### Comment by abdullah 2018-04-27 10:23
Not really. I think the reporter is asking for improving the strings of the error messages that already exist. I am not sure if this is reasonable for all cases, as there may be many combinations.

I am moving this to Sketcher, as it is not really GCS related...

maxwxyz, 2024-02-25:
@PaddleStroke do you think this is still relevant?

PaddleStroke, 2024-02-25:
I'm unsure. The situation evolved significantly since this issue was
opened. Also it is a bit broad as it's asking to improve messages but not
which exactly. I think printing what the user has selected on error is
overkill.
Imo it can be closed.

closed 2024-02-25 by maxwxyz