FreeCAD/FreeCAD #5578: DOS line endings makes applying patches on Linux really hard

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1352

* **Reporter:** jobermayr
* **Date submitted:** 1/11/2014
* **FreeCAD version:** trunk
* **Category:** Bug
* **Status:** new
* **Tags:** #tobeclosed

# Original report text

I must apply patches on pmbs via "patch --merge -p1 <$PATCH" instead of just "patch -p1 <$PATCH" due to silly DOS line endings:
https://pmbs.links2linux.de/package/show?package=FreeCAD&project=Extra

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Platform:** Linux
* **Updated:** 11/5/2021

# Discussion from Mantis ticket

### Comment by jobermayr 2014-01-12 15:10
Btw. it seems to be an easy task:
https://help.github.com/articles/dealing-with-line-endings

---

### Comment by shoogen 2014-01-12 18:28
Did i get your right? You proposed to normalize the whole repo to LF line endings?
http://stackoverflow.com/questions/1510798/trying-to-fix-line-endings-with-git-filter-branch-but-having-no-luck/1511273#1511273

---

### Comment by jobermayr 2014-01-12 19:14
Please try to apply https://pmbs.links2linux.de/package/view_file?expand=1&file=0001-Add-support-for-different-libdir.patch&package=FreeCAD&project=Extra

1. git am $file # fails
2. git reset --hard origin/master
3. patch -p1 <$file # fails
4. git reset --hard origin/master
5. patch --merge -p1 <$file # works
6. git commit -a --date="Mon, 2 Dec 2013 22:21:02 +0100" --author="Johannes Obermayr <johannesobermayr@gmx.de>" -m "Add support for different libdir."
7. git format-patch origin/master
8. git reset --hard origin/master
9. git am 0001-Add-support-for-different-libdir.patch # fails because of silly DOS line endings

So if you have another fix please let it know ...

---

### Comment by shoogen 2014-01-12 20:07
i don't even get to download the patch. <summary>Authentication required</summary>

---

### Comment by shoogen 2014-01-12 20:20
You could ask the packman build service guys to invoke "git am" with the parameter --keep-cr
or "diff" with "--binary"

---

### Comment by jobermayr 2014-01-12 21:16
Since I stepped in and am the de facto maintainer of FreeCAD on Packman I can tell you that %patch macro from openSUSE (main distro) applies patches via "patch $level -fuzz=0" also Fedora and some other RPM based Linux distros are doing it this way.

JFYI: None of the other packages on build.opensuse.org and pmbs.links2linux.de makes problems in the way FreeCAD makes (not FHS conform, RPATH issues, silly DOS line endings, ...)
Hacks and workarounds as well as hacks and workarounds to get them applied are needed to fulfill minimal requirements to get it working and accepted ...
Proper fixing is not possible in just a few minutes because jumping between c/c++ and python is a hell ...

Fixing OCE to our needs was a laugher:
https://github.com/tpaviot/oce/issues/created_by/jobermayr?state=closed
https://pmbs.links2linux.de/package/view_file?expand=1&file=0002-OpenGl_GlCore12.hxx-undef-GL_VERSION_x_y-in-all-case.patch&package=oce&project=Extra

Btw. the download problem is (WONTFIX on pmbs):
https://github.com/openSUSE/open-build-service/issues/458
Maybe you can copy & paste it?

---

### Comment by shoogen 2014-01-13 16:57
IMHO adding a single commit to our repo that will remove all Carriage Returns has some drawbacks. 
The most important is that it will render the "git blame" command useless and that will make the repo bigger.
I would prefer to orphan the commit that will remove all the CRs. But would break all branches that have diverged from master.
But I am just an ordinary contributer.
Maybe we can just remove the CRs in all places touched by your patches, so they will apply cleanly. This won't solve the problem, as it would have to be repeated for every patch you add (in the future). But it will provide time to think about a suitable solutions, whilst allowing you to continue using PMBS.
And as this bug flagged as 'urgend' I think that it might be worth it.

---

### Comment by shoogen 2014-01-19 13:26
this ticket is one week old. I see it is still major issue, but don't think that it is urgent anymore, sine there seems to be a workaround in place.

BTW: when debugging line end ending issues, i would usually avoid copy & paste.
https://github.com/openSUSE/open-build-service/issues/458
openSUSE 13.1 was out the day you wrote that

---

### Comment by Kunda1 2017-01-10 16:25
shoogen Does this workaround need to be documented somewhere and then we can close this ticket?

---

### Comment by Kunda1 2017-01-17 04:05
Related forum threads:
http://forum.freecadweb.org/viewtopic.php?t=15443
http://forum.freecadweb.org/viewtopic.php?t=19866 <-Active

---

### Comment by blacey 2017-03-03 22:06
To me, this is something that wmayer or yorik should do because it will create a broad swath commit across the repo.  I think with a little bit of communication, this can, and should, be done quite easily.

---

### Comment by ian.rees 2017-03-03 22:50
I think that **if** we're going to do a massive line-ending-change (my vote is to not to), it would be best to do it using something like filter-branch that rewrites history so we don't have a commit that touches every line in the project.  We also need to have a process in place to make sure that the line endings stay consistent.

It will be a pain either way, but with the rewrite option, pain goes away once everyone's dev branches are based off the "new" history.  Otherwise, it's always going to be more difficult to understand history from before the line ending fix.

Someone would need to come up with the process for moving the project to the new line ending style, and to handle a few common scenarios (like how to merge in branches that were made from the history with the wrong line endings).

---

### Comment by yorik 2017-03-04 00:22
Won't it be an endless struggle, that even if we unify everything, very soon new commits will be added with "wrong" file endings again and we'll need to do this forever?

---

### Comment by ian.rees 2017-03-04 02:50
yorik - I think you're right.  To keep line endings nice, we would also need a system to screen changes.  That system might be nice, because it would help avoid individual files with both types of line endings.

As I understand the original issue above, unifying line endings to be "Unix" type isn't really going to solve the problem.  The issue seems to be that the patches have line endings that aren't consistent with the file being patched.  So, if we standardise on Unix then it makes creating good patches easier on Linux/MacOS, but harder on Windows.

---

### Comment by blacey 2017-03-04 06:38
The "endless struggle" could be averted through two methods:  1) add the line ending config to the .gitattributes file and 2) Add checks to Travis. 

If everything is configured properly, all operating systems should conform to the line ending config while modifying files within the FreeCAD git repo.  If someone goes astray, Travis can catch it while validating the Pull Request.

---

### Comment by Kunda1 2017-10-19 13:12
Linking #3206

---

### Comment by Kunda1 2019-09-05 20:27
wmayer do you think this ticket still has merit ?

---

### Comment by wmayer 2019-09-10 11:50
In >99% of all cases we get contributions via GH (or Gitlab) and very rarely real patch files. So, patches with incompatible line endings is not a big issue any more.

But I would still leave this open because also on GH it happens from time to time that the line endings of a file has completely changed while only a single line has been modified by the author. This complicates the review, then.

---

### Comment by chennes 2021-09-15 16:06
wmayer, since there is no action to be taken on anyone's part, except to make sure your PRs don't mess up all the line endings, I suggest we close this. It's not a bug at this point, as much as "how to be a good open source developer" advice that would best be suited to a statement on the Wiki.

---

### Comment by wmayer 2021-09-17 14:57
We haven't had any serious issues with line endings in the last couple of years or got patches that couldn't be merged easily. Also if a PR would change all line endings I found the trick to add ?w=1 to the url to ignore white spaces changes when reviewing the code.

yorik what do you think?

---

### Comment by ferdymercury 2021-11-05 16:43
Suggestion: one could mark the "big" commit changing all the line endings in a .git-blame-ignore-revs file, so that it can be easily 'hidden' from history, but does not make a mess with rewriting history and branches. This is also useful if one wants to use auto-formatters to reformat old code into a specific consistent style.

See:
https://github.community/t/support-ignore-revs-file-in-githubs-blame-view/3256
https://michaelheap.com/git-ignore-rev/
https://gitlab.com/gitlab-org/gitlab/-/issues/31423

adrianinsaval, 2022-02-11:
IMO we should change line endings once and for all and add to the .git-blame-ignore-revs as suggested or close this issue and forget about line endings

luzpaz, 2022-12-03:
@donovaly can we do this first thing after the next release (and add it to the `.git-blame-ignore-revs` file subsequently) ?

luzpaz, 2023-03-13:
What FreeCAD release version do we want to implement removing DOS line endings from FC source code ?   
Are there any large milestones we want to wait for until we do this? (AKA toponaming)

donovaly, 2023-03-13:
I fully agree with @wwmayer comment here from 2021 that there is no issue anymore -> closing

luzpaz, 2023-03-13:
It still is confusing/annoying to submit PRs and see:!

1. CI indicate the file has non Unix line endings  
[image]

2. `^M` at the end of each line when viewing in Vim or IDEs that show line endings

For uniformity sake, wouldn't it make sense to just take care of this once and for all ?

donovaly, 2023-03-13:
> This can be removed. But also this situation is quite boring when doing a git diff (even if there is an option to ignore it). So why not moving once all files to one standard then fail CI when a faulty file is pushed?

Because this increase the time you need to work for FreeCAD.  mean whenever I edit something, it might be that Visual Studio, Inkscape, jEdit or whatever program changes the line endings. But this is no issue with Git so I save time that I don't need to check for this.

As you know me ;-), I am eager to save time and keep the level to to contribute to FC.

donovaly, 2023-03-13:
> when doing a git diff (even if there is an option to ignore it).

Not here. I use the option to ignore the line endings (commit as is) and on making a diff I don't get a diff for line endings. I use Git 2.39.2 and the latest TortoiseGit as frontend.

luzpaz, 2023-03-13:
> I use the option to ignore the line endings  

Not everyone does. Just being mindful towards contributors. It's a simple fix and easily hidden in [`.git-blame-ignore-revs`](https://github.com/FreeCAD/FreeCAD/blob/master/.git-blame-ignore-revs) for the devs when searching git history.

donovaly, 2023-03-13:
> > I use the option to ignore the line endings
> 
> Not everyone does.

It is the default when I install Git.

However, let's please stay it as it is. We have very rarely problems and people should not need to check for line endings but focus on the code and Git does the job in the background. So as we work already, is apparently fine.
We have here one year-old report and start a debate. But we should only do this when several people have real issues. Before you re-activated the discussion here, I did not see anything the last years from the different code contributors.

luzpaz, 2023-03-15:
Theoretically we will need to do #6415 which will overwrite the whole FC codebase anyway. Might as well do it in conjunction with migrating away from dos line endings as well...

luzpaz, 2023-03-17:
It's a leap but not a large leap. Hiding the commit(s) in [.git-blame-ignore-revs](https://github.com/FreeCAD/FreeCAD/blob/master/.git-blame-ignore-revs) should make the change harmless as github will not display those commits.

wwmayer, 2023-03-17:
Should we reopen this ticket?

0penBrain, 2023-03-17:
> Should we reopen this ticket?

Probably yes. At the end I think we need to have first a consensus about when we want apply all this "cosmetics" changes (vs. dev timeline) then we have to choose which one to bring in.

luzpaz, 2023-03-17:
> Should we reopen this ticket?

Reasons to consider: 
* code uniformity throughout the project
* diminishes confusion for devs that may submit patches (outside of github workflow)
* diminishes confusion for devs that use non-modern code editors
* future friendly 

I'd also suggest re-enabling check for line endings 5ac308e8a65c7022ef851463d8ab5e1ba4b5edbe

donovaly, 2023-03-17:
> Should we reopen this ticket?

Please not. I mean developing FC is already complex. I am super happy that I don't have to care also ofe line endings. For years now it was no problem, and now it should be on?
I really don't understand why we can't keep it as it already is. 

I want to focus on the actual development not be forced to be punished to use Windows.

When uniformication, why Linux file endings and not uniformly DOS (the OS used the most by the users)?
If we would to it that way, many developers would see how annoying and time-consuming it is to also take care of line endings.

luzpaz, 2023-03-18:
* The only time consuming issue is this debate and waiting for a moment to change line endings in the code.
* Windows users aren't forced to use linux line endings. 
* Windows users aren't being punished.

luzpaz, 2023-04-02:
Another example https://github.com/FreeCAD/FreeCAD/pull/9149#issuecomment-1493213500

wwmayer, 2023-04-02:
Since there are still some ongoing discussions I think it's best to reopen it now.

xtemp09, 2023-04-02:
I'd add that the files should be converted into UTF-8.

wwmayer, 2023-04-02:
Since we plan to enable pre-commit hooks the maintainers have to make sure that all their modules are formatted according to clang-format and Black rules. So, we could use this opportunity to convert into UTF-8 and UNIX-style line endings.

adrianinsaval, 2023-04-02:
Ensuring correct line endings can be apart of the commit hooks too right?

0penBrain, 2023-04-02:
> Ensuring uniform line endings can be part of the pre commit hooks too right?

Even converting to specific line endings can be in the hook. ;)

chennes, 2023-04-02:
Yes, I have added that hook now, it will keep everything at UNIX line endings (of course, right now only in the Addon Manager, but eventually everywhere).

berndhahnebach, 2023-04-03:
I do not know if you guys are aware of this ... Does it needs changes because of the new hook? https://github.com/FreeCAD/FreeCAD/blob/62bc5c741147a118a52b52aab070e85939c78f90/src/Mod/.gitattributes

chennes, 2023-04-03:
@berndhahnebach normalizing is disabled for all Mods, isn't it? So the new hook does ignore this file, it's a sort of "side-channel" rather than using git itself to do the work. I suppose we should ultimately delete this one.

berndhahnebach, 2023-04-03:
> @berndhahnebach normalizing is disabled for all Mods, isn't it? 

No it is activated for code files of these modules ...
# AddonManager/** -text
# Arch/** -text
# Complete/** -text
# Draft/** -text
# Idf/** -text
# Fem/** -text
# Material/** -text
# OpenSCAD/** -text
# Path/** -text
# Show/** -text
# Tux/** -text

just read the text in the file I have posted the link.

chennes, 2023-04-03:
Oh, I see, I misunderstood how it worked, thanks. So then that part of the hook won't matter for those Mods because their endings are already normalized. Once this process is complete we can probably remove this file, since the precommit hook will be doing this work. Or, we can turn off that precommit hook and let git do this to all mods.

luzpaz, 2023-08-13:
@chennes where are we on pre-commit hooks ?

chennes, 2023-08-13:
The Maintainers responsible for the various parts of FreeCAD need to work with their development teams to decide when and how to apply the pre-commit hooks to the parts of the codebase they are responsible for. It's so far done for AddonManager and Test, for example.

luzpaz, 2023-08-13:
- [X] Addon Manager
- [X] Arch (@yorikvanhavre)
- [ ] Draft 
- [ ] FEM
- [ ] Part
- [ ] PartDesign
- [ ] Path
- [ ] Sketcher
- [X] Test

berniev, 2023-08-13:
Do cmake files need consideration too?

chennes, 2023-08-13:
cMake files are considered by the basic tools (line ending, whitespace, newline-at-end), but there isn't a first-party auto formatter for them. Do you have any experience with https://github.com/cheshirekow/cmake-format-precommit (or any other third-party auto formatter)? We have auto format for C++ and Python, so having the same for cMake would make sense to me.

berniev, 2023-08-13:
I have recently discovered that CLion does a magnificent job formatting cmake files, including fixing case, setting indents etc. I have seen the mentioned code referenced but when I looked at the GitHub wasn't sure it was current. I'm not sure, but is CLion using clang-format for cmake files?

chennes, 2023-08-13:
I don't think so: according to https://clang.llvm.org/docs/ClangFormat.html, 

> [clang-format] can be used to format C/C++/Java/JavaScript/JSON/Objective-C/Protobuf/C# code

If there's a way to activate clang-format such that it checks cMake files, I can't figure out how. But it does seem like there are some public cMake formatters out there. I have no idea how good/flexible any of them are, though.

berniev, 2023-08-13:
As an initial step I'd like to see the whole cmake directory formatted. In CLion its just a couple of clicks. Some macro and functions names will have to be manually updated to snake case but that's no big deal. At least that would establish a solid starting point. At the moment editing cmake files is a PITA.

The cheshire.. thing seems to be inactive. There is an issues raised in February questioning status and has not seen a reply.

chennes, 2023-08-13:
OK, if you want to submit a PR that bulk-reformats the files in the cMake directory I think we can start there. We can also add that subdirectory to the pre-commit hook setup so at least those basic formatting things get caught.

luzpaz, 2023-10-24:
@chennes what's the verdict on this ticket now that we have pre-commit ?

abdullahtahiriyo, 2023-10-25:
@luzpaz 

Veredict: We should erect a statue of chennes for having introduced pre-commit.

luzpaz, 2024-09-08:
@wwmayer jfyi, if you address this issue in the bug pre-commit commit, please add this ticket to the commit. TIA!

hzeller, 2024-09-15:
Now that 1.0rc1 got more peoples' attention (including mine :) ) I looked around in the source-code to maybe contribute.

The mix of files with `LF` and `CRLF` makes it really hard to work with files and changes and feels like having to walk on egg-shells to not mess with the line-endings in particular files. It would be really nice to see this (mostly) resolved before 1.0.

## Auto-fix changes whenever a file is modified and updated

I think a good strategy would be to use `.gitattributes`  which can be configured to auto-convert to a canonical LF format:

https://docs.github.com/en/get-started/getting-started-with-git/configuring-git-to-handle-line-endings#per-repository-settings

Probably something like this global `.gitattributes` in the root of the repo.

```
*.cpp text
*.h   text
*.py  text
*.cxx text
*.hxx text
*.inl text
```

These file are then auto-converted to the local platform newline format on check-out (or, we can have that `text eol=lf` to always be `LF`, which is an increasingly preferred way of doing things even on most Windows developers).

### Caveats
It might make require to have folks who already have checked out a working copy be a bit more careful if they pick up the `.githattributes` which then changes all the files at once on next `pull`. This should not be a big deal, but I can imagine corner-cases in which it might mess with open pull requests as that might consider all non-modified files in that PR to look like they changed.

So: probably not an issue in real-life, as there is good tool-support, but there can always be accidental situations where things get discombobulated. So I suggest a...

## First Preparation Step
To reduce the impact, I suggest to have a first step: to convert all the files currently _not_ open in any pull requests first, and submit to the main repo, so that the baseline count of files with `LF` is larger.
Only then add the `.gitattributes`.

## TL;DR
I am happy to do the leg-work and move this Issue forward: finding all the files that are currently not open in any pull requests and send a bulk PR to change newline changes in `src/**`. However, since I have not contributed to FreeCAD (yet), it would be good to know if the maintainers would accept such a PR before I start.

JohnOCFII, 2024-09-15:
I'm sure one of the maintainers will be along with detailed input, but in the meantime, wanted to make sure you've seen this document: https://freecad.github.io/DevelopersHandbook/

hzeller, 2024-09-16:
These are the files that need a `CRLF` -> `LF` conversion and are currently _not_ mentioned in any pull request, so would not conflict when modified in a bulk update (1614 out of 1975 in need).

[freecad-files-need-newline-no-pr-conflict.txt](https://github.com/user-attachments/files/17007802/freecad-files-need-newline-no-pr-conflict.txt)

created with:

```bash
#!/usr/bin/env bash
#
# Needs shell utilities find, file, sort, awk, gh, jq, wc, join
#
# Determine files that need and can be converted from CRLF to LF as they are
# not conflicting with any open PR.

FILES_OPEN_IN_PR=/tmp/freecad-open-pr-files.txt
FILES_NEED_NEWLINE=/tmp/freecad-files-need-newline.txt

# Files that need newline and don't conflict with any open PR
FILES_NON_CONFLICT=/tmp/freecad-files-need-newline-no-pr-conflict.txt

# git-open-pr-files in https://github.com/hzeller/dev-tools
git-open-pr-files FreeCAD FreeCAD > "${FILES_OPEN_IN_PR}"

file $(find src -name "*.cpp" -o -name "*.h" \
	    -o -name "*.hxx" -o -name "*.cxx" \
	    -o -name "*.inl" -o -name "*.py") \
    | grep CRLF | awk -F: '{print $1}' \
    | sort > "${FILES_NEED_NEWLINE}"

# Files that need fixing, but are not also open
join -v1 "${FILES_NEED_NEWLINE}" "${FILES_OPEN_IN_PR}" > "${FILES_NON_CONFLICT}"

wc --total=never -l "${FILES_NEED_NEWLINE}" "${FILES_NON_CONFLICT}"
```

```
  1975 /tmp/freecad-files-need-newline.txt
  1614 /tmp/freecad-files-need-newline-no-pr-conflict.txt
```

There could be probably multiple bulk updates over a few weeks while PRs are merged until the remaining list of files is so small that adding the `.gitattributes` will be a good next step.

hzeller, 2024-09-16:
Thanks @JohnOCFII , the https://freecad.github.io/DevelopersHandbook/ is a good resource.