FreeCAD/FreeCAD #5580: Allow editing of multiple sketches at the same time

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1391

* **Reporter:** normandc
* **Date submitted:** 2/7/2014
* **FreeCAD version:** 0.14
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Currently, it is not possible to edit two sketches at the same time. To edit a second sketch, the user has to close the first sketch.

It would be very useful to allow editing of more than one sketch at the same time.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by mrlukeparry 2014-02-08 00:34
I am just wondering if this is just to do with sketcer or might be related in general to ViewProvider::setEdit()?

If I remember there is specific dialogue warning if you try to open another sketch editor.

luzpaz, 2022-02-23:
Is this AKA Multiple viewports open simultaneously? Sort of how @microelly2 used to do it..
[image]

maxwxyz, 2024-02-21:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11879

closed 2024-02-21 by maxwxyz