FreeCAD/FreeCAD #5582: Constraints creation "on the fly"

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1416

* **Reporter:** Pauvres_honteux
* **Date submitted:** 2/16/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** constraints, Solver

# Original report text

Hi, I have a wish of getting a constraints creation tool that consists/contains of all possible constraints commands avaible and starts running after double clicking a general constraints icon.

Similar principle behavior as in a well known french cad program.

I.E. after this general constraints tool has started you just click on the two lines/points you want to constrain and the following behavior is the same as now with the addition of a right click meny where one can select submatters to that specific constraint if applicable. And here's the beauty in it: you do not need to move your cursor up to and click the constraint icon for every constraint you want to create. This saves your hand from getting over strained and speeds up your work - a lot.

To stop it running one could click the general constraints icon again or press ESC.

Regards Pauvres_honteux

# Additional information

Had this in the "Open discussion" before but it was apparently the wrong place.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 6/9/2021

# Discussion from Mantis ticket

### Comment by Pauvres_honteux 2014-05-18 07:01
One more who finds this principle behavior to be of interest:
http://www.freecadweb.org/tracker/view.php?id=1518

---

### Comment by Kunda1 2017-05-22 13:45
Abdullah OP seems to be referencing a ticket you worked on #1518 as similar. Can you weigh in?

---

### Comment by abdullah 2017-05-22 22:00
Well, I did the Continuous geometry mode you refer to. Jxnd did the continuous constraint creation mode (you click the constraint icon and now you select elements one after the other to which you want to apply the constraint, or pairs, or triplets depending on the constraint).

I am not sure how that "french cad program" works. Apparently the reporter is asking for a generalisation of what jxnd did. 

I am not sure I understand the part of "a general constraints tool". If it is general, how does it know which constraint you want to apply? by guessing? The right click menu initially I thought was to change from one constraint to another, or select from a computer generated guess list. However, it says submatters, so it seems some kind fine tunning?

A while ago I had an idea (never implemented, it will be in the forum somewhere) called "constrainator" which was guessing constraints based on the relative position of the geometry. This ticket reminded me of that.

All in all. There is potential for improving constraint creation. Reducing the strain should be one of the main ergonomics goal of any CAD software, as a CAD user spends tons of hours in front of the computer, and it is well known that this creates all kind of health problems. 

Now, I would ask the reporter to better define what he expects from this feature request. With the current explanation there are little odds a developer will take it. Even lower that the result will match the expectations.

---

### Comment by Kunda1 2017-05-23 04:38
Pauvres_honteux ping

---

### Comment by Pauvres_honteux 2017-06-17 07:35
Tried to make a movie but it got enormous in byte size, so I made a series of pictures instead.
Could not bring the whole screen in on the pictures due to bytes limitation, but I guess icon appearance and double clicking isn't that interesting anyway in this context?
Started the sketching tool "Polyline with arcs" to make the following:
While clicking along in the sketch I can make an arc by "click-drag-click" as in frame 10 -> 12. Since "auto constraint" is active it also creates a tangent constraint.
The blue lines and symbols are "running constraints", which **suggests** constraints at the tip of the pointer. As you can see in frame 13 it **suggests** a tangency constraint which I accept by left clicking once.
Frame 40 -> 43 shows an constraint which creates a "contradictory" constraint, had to be deleted. Was done by right clicking and selecting "Delete", i.e. **I didn't have to stop the "constraining mode" for the deletion.**
The green constraints where done by double clicking the "constraint button" once. After that you just "constrain away" !
To temporarily deactivate "running constraints" you press "shift" button.
To create just one constraint you single click the constraint button.
Extra: 
[url=https://forum.freecadweb.org/viewtopic.php?f=34&t=18175&p=150282&hilit=running%2A#p150282]Better selection system for constraints in Sketcher
[/url]

---

### Comment by Pauvres_honteux 2017-06-17 07:54
This "Tracker" platform adds "https://www.freecadweb.org/tracker/" in front of my inserted and intended link!
It est, the link doesn't work at all !
On top of the above I can't edit what I've written once it's sent!
I'm using openSUSE Leap 42.1 with Xfce 4.12 and Firefox 52.1.1 (64-bitars).

Anyway, copy the link "Better selection system for constraints in Sketcher", paste it in your browsers address row and remove "https://www.freecadweb.org/tracker/", then execute it.

---

### Comment by Kunda1 2017-06-17 11:32
Sorry you're wrestling with the tracker. I fixed the URL issue all it needed was https:// as a prefix. I'll submit an upstream bug report about it. In regards to editing previous entries the tracker doesn't give that permission to 'reporters' maybe we should change that? (Ping  kkremitzki )
Re: large size attachments you can always use an3rd party to host such a file. For movies some people use youtube. HTH

---

### Comment by kkremitzki 2017-06-20 02:17
Kunda1 Pauvres_honteux Reporters should now be able to update issues.

---

### Comment by Pauvres_honteux 2017-06-20 05:11
kkremitzki, thank you so much for this editing possibility!
This really opens up for reducing tracker bloating.

PS. I removed a couple of my posts not adding anything to the issue. DS.

---

### Comment by abdullah 2018-05-21 06:15
Thanks for the pictures. I have understood what you mean. 

Some of the concepts are similar to an old idea called "Constrinator", which was never implemented. I intended to start developing parts of that tool this year. 

I will consider your input when giving shape to that new tool.

---

### Comment by Murmele 2018-05-21 17:35
I created a small clip which shows the dimensioning in catia:
[Youtube](https://youtu.be/bvCXvQ1rMoo)

adrianinsaval, 2022-02-11:
#5407 adresses this request

adrianinsaval, 2022-07-19:
Original "discussion" for this issue: https://forum.freecadweb.org/viewtopic.php?f=8&t=5711&p=46304 although this discussion by paddle is more relevant: https://forum.freecadweb.org/viewtopic.php?f=9&t=65521

howie-j, 2023-08-30:
Solved by #9810.

closed 2023-08-30 by adrianinsaval