FreeCAD/FreeCAD #5583: Half width tangential lines

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1417

* **Reporter:** Pauvres_honteux
* **Date submitted:** 2/16/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #pending, colors, GUI, lines

# Original report text

Hi, I would like to see tangential lines width half the size of boarder lines and in a bleaker color. This makes it easier on the eyes and you can emediatly see if two surfaces are tangential or not - speeds up work.

Regards Pauvres_honteux

# Additional information

Had this in the "Open discussion" before but it was apparently the wrong place.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Pauvres_honteux 2014-02-16 19:58
Sorry for not being clear enough. With tangential lines I mean the lines that are visible on a surface where, for example, a flat surface pass over to a curved surface like in a radius between two perpendicular surfaces. When one see that passage you would immediately get a visual confirmation on its state: sharp (boarder line width), tangential (half boarder line width), or curve smooth passage (no line at all).

Example:
The three different basic passages - http://s3files.core77.com/blog/images/3surface.jpg 

The black line equals C0 (not tangent and sharp), the "half width bleak colored" light grey line equals C1 (tangent), the gap will eventually be filled by a curvature correct surface and its boarders limit will no be visible, i.e. C2 - http://architosh.com/wp-content/uploads/2013/09/03_evolve_chair_surface_blend.jpg

Tangential lines with no difference to boarder lines - http://www.inceptra.com/images/newsletters/newletter6/catia_14.jpg

---

### Comment by mdinger 2014-05-18 17:49
I find this explanation confusing.  Does Rhino do tangency this way?  Here is my interpretation of your request referring to each C elements from the first picture?  It won't agree with the pictures.  I have no idea what they are trying to explain.

C1-3 picture:
1. Circle should be black because it is *not* tangent.
2. Line or Curve should be faint because it *is* tangent

---

### Comment by Pauvres_honteux 2014-05-19 20:50
Hi guys, pictures were done in Catia. Don't ask me why I went over the top with all the tangent lines and tangent circle and what not. Maybe I had the first picture I made a link to in mind...
Anyway, you seem to understand what I originally proposed and that's what counts.
In V5 you can also turn off all internal tangent lines. While this makes any non tangent line really stand out it's hard to see how the surfaces really looks like.
To summarize it a bit I'd say an really easy option for switching between different viewing behaviour is high on my whish list.

---

### Comment by mdinger 2014-05-24 00:08
Solidworks supports this.  See `solidworks_construction_tangent.png`.

Note: It can be difficult to tell that they are construction linetype sometimes because of aliasing.  Line thickness does not appear to have changed.

---

### Comment by Pauvres_honteux 2017-01-15 07:06
Hi, Kunda1 !
Sure, here it is:

http://forum.freecadweb.org/viewtopic.php?f=8&t=5712

---

### Comment by Pauvres_honteux 2017-06-18 09:02
Shame on thy giving up...
Hopefully these pictures are clear enough?
To clarify this once and for all: this issue is about **the line seen between two surfaces in contact with each other**. In CATIA there's an exception: curvature transition between surfaces does not show any line, but it would be beneficial to have another line type for that.

Kunda1, what do you think about cleaning up this thread and remove all misunderstood/irrelevant stuff? I think the links I originally posted can be removed as well.

---

### Comment by Pauvres_honteux 2017-06-20 05:33
Kunda1 , could you please remove the pictures from post 009442, which includes the picture above it?

kkremitzki , would it be possible to let us users remove/change/edit the pictures we've uploaded as well?

---

### Comment by kkremitzki 2017-06-20 05:50
Pauvres_honteux actually I need to revert the change because it allows any 'reporter' user to edit others' issues. Standby

---

### Comment by kkremitzki 2017-06-20 06:04
Pauvres_honteux Kunda1 from http://mantisbt.org/manual/#admin.lifecycle.workflow.thresholds it doesn't seem like there's any granularity to

[quote]Update an issue - The access levels that are allowed to update the header information of an issue. [/quote]

and therefore no option for reporters to edit their own issue headers. I also don't see any option for deleting one's attachments

---

### Comment by Kunda1 2017-06-22 02:10
Pauvres_honteux tell me what to remove and I'll make it happen

---

### Comment by Pauvres_honteux 2017-06-22 04:33
Kunda1 , thank you for your kind offering, please see post 0009466.
And if you feel adventurous; all the other posts just adding to the confusion:
0004227
0004696
0004697
0004698
0004699
0004700
0004703
0004704
0004705
0007741

---

### Comment by Kunda1 2017-06-22 10:57
Pauvres_honteux Done! ;)

---

### Comment by Pauvres_honteux 2020-10-22 11:26
My pictures seems to have vanished!?!?
Oooh well, I'll add a couple of new ones.

Picture 51.png shows the tangential lines in full width and black.
Picture 52.png shows the tangential lines in half width and grey.

And please do not delete my added files from now on.

---

### Comment by Kunda1 2020-10-22 15:48
They weren't deleted purposely. It may have been a broken by an update or something. There are other tickets with missing attachments we all.

maxwxyz, 2023-05-06:
I would like to add an additional option for this feature request: To hide smooth edges completely. I have created a forum post about this: [View: Shaded with Hard Edges](https://forum.freecad.org/viewtopic.php?p=680270) which would result in a shaded view combined with edges but without smooth edges.

[image]

Especially for complex cast parts like housings or when working with surfaces it is helpful to identify hard vs smooth edges. It allows to quickly see hard edges for e.g. a spotface or helps to see the continuity of a surface. Also, in complex parts, too many edges are more disturbing than helping but often you need to select or use the sharp edges for new features or as references.

In CATIA and NX, you can use a specific composite rendering style to shade the surfaces of an object with its edges, except the smooth ones. A smooth edge is an edge that does not break the continuity of the geometry, contrary to a sharp edge.
Different render styles are shown [here](https://www.3ds.com/fileadmin/3dxml/dircopy_official/PlrUserMap/plr-t-Player-RenderingStyles.htm)

adrianinsaval, 2023-06-22:
closing in favor of #9636

closed 2023-06-22 by adrianinsaval