FreeCAD/FreeCAD #5584: Sketcher: Break a Line and Add a Point onto a Line

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1436

* **Reporter:** bejant
* **Date submitted:** 2/23/2014
* **FreeCAD version:** 0.13
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

A forum user requested two new features which seem useful:
1.  Break a line into two segments, leaving a gap where the user clicked between the two new segments.  I'll add that the default gap should be user selectable.    
2.  Add a point onto a line.
From the following forum thread:
http://forum.freecadweb.org/viewtopic.php?f=8&t=5928#p47176

# Additional information

Feature request #1 reminds me of a little lisp routine named brk.lsp that I used to use daily in AutoCAD.  I did a web search and believe the following code is the proper routine, maybe it will be of help (or maybe not).

(defun c:brk(/ os pt1 pt2) 
  (setvar "CMDECHO" 0) 
  (setq os (getvar "osmode")) 
  (setvar "osmode" 512) 
  (setq pt1 (getpoint "\nPick object to be broken : ")) 
  (setvar "osmode" 33) 
  (setq pt2 (getpoint "\nBreak point : ")) 
  (setvar "osmode" 0) 
  (command "break" pt1 "f" pt2 "@") 
  (setvar "osmode" os) 
  (princ) 
)    

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 5/25/2021

# Discussion from Mantis ticket

### Comment by aviket 2021-05-25 10:42
This feature is already partially present in draft module of 0.19

AjinkyaDahale, 2023-01-10:
So do I understand correctly that you want a tool like the "trim" tool that instead splits the edge at the intersection? What do you mean by "little gap"? A minimal demo of this interface would be best.

luzpaz, 2023-01-10:
@AjinkyaDahale i can't access the forums ATM can PM [bejant](https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=1940) to come participate in this discussion ?

AjinkyaDahale, 2023-01-10:
@luzpaz GeneFC pointed out that bejant hasn't been online since 2020!

luzpaz, 2023-01-11:
bummer :disappointed:

Roy-043, 2023-07-15:
The Sketcher_Split command fulfills request #2. Using it twice on the same line and deleting the middle section would fulfill request #1. IMO this can be closed.

closed 2023-12-09 by abdullahtahiriyo