FreeCAD/FreeCAD #5585: Save CAD data as text (JSON) for version control

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1558

* **Reporter:** petebachant
* **Date submitted:** 5/19/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

This may be too drastic of a change, but it would be really great to be able to use Git, or some other text-based version control tool with CAD data. 

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by petebachant 2014-05-19 21:02
I just found out that the FreeCAD format is actually a compressed archive containing XML files. Maybe an option to load/save uncompressed would suffice.

---

### Comment by normandc 2014-05-22 01:33
Revision control is a long-term goal of the FreeCAD developers, see http://www.freecadweb.org/wiki/index.php?title=Assembly_project#Revision_control

---

### Comment by shoogen 2014-05-22 07:17
We save zip files, which contain different type of data.
What you could do is to extract the FCStd file to a directory and add the whole directory to your version control.

Putting everything into one text based file, (XML, JSON...) makes not much sense if we had to embed our BREP files.

We could try to use OCAF to get an XML representation of shapes.

---

### Comment by shoogen 2014-09-22 13:24
It seems that OCAF XML just includes she Shapes as CDATA

---

### Comment by Kunda1 2018-04-28 02:48
Related: https://github.com/reox/FreeCAD_gitproject

---

### Comment by freelibre 2019-08-11 05:57
Version control with git in a gitlab or github project repo would be a great feature to improve open hardware projects team members collaboration.

The most usefull feature would be to be able to see live versions & branches of the different parts designed in a project by the different project members.

That's a feature Onshape has implemented and a short demo is available there:
https://www.youtube.com/watch?v=ggL9IQSo4VI

mmuman, 2023-02-04:
Just arriving to FreeCAD from OpenSCAD which has some limitations but is a breeze to version in git, and I don't even feel like designing anything until I know how I can properly version things.

There's at least two incompatible ways I found:
- just putting the zip directly into git, optionally with some difftool sugar with [WebTools_Git](https://wiki.freecad.org/WebTools_Git#Limitations), which I don't find acceptable, I mean even LibreOffice has a Flattened version of their formats (FODT),
- using [a local smudge filter](https://blog.lambda.cx/posts/freecad-and-git/), and version the dumped output, but then anyone cloning the repo will need to configure it manually to get a usable file,
- and [someone mentioned me](https://tooter.social/@Nic/109805002368310172) realthunder's branch (but which one, there's 224 of them?), where you can save as unzipped folder. But if it's never merged upstream then people checking the repo will also need to install the patched version.

So there's 3 ways equally unusable to me right now. :shrug: 

While it's possible to use any of these locally and later convert the repo with something like filter-branch, once a repo becomes public it kinda becomes set in stone, as rewriting public git history is frowned upon.

Is there any plan for advancing on this?

Would contribution to WebTools to handle the smudge way be accepted?

Is the folder-based format decided yet? I wonder if a smudge filter could actually use the same format meanwhile…

mmuman, 2023-02-04:
And I just now find the relevant forum threads, for the record:
https://forum.freecad.org/viewtopic.php?p=182805
https://forum.freecad.org/viewtopic.php?f=10&t=38353

chennes, 2023-02-05:
It never hurts to submit a PR with something that works, so we can have a more concrete discussion. That said, have you looked at a diff between two versions of a FreeCAD file? My feeling is that the result is not especially useful. Where with OpenSCAD you are storing commands to create geometry, in FreeCAD you are storing the geometry itself (as BREP files). So while technically they are text files, a diff between them can't really be treated the same way. If you want a more OpenSCAD-like experience I think you'd need to write (or record) a macro that creates your geometry, and version that instead.

mmuman, 2023-02-06:
Yes I got that from reading the forum threads, hopefully the directory-save work can improve on that, but I don't expect to be able to automatically merge commits anyway.

As for macros, seeing both Python plugin codes and videos about how not to mess up the hierarchy depending on which workbench you start using, I actually thought of writing the python code and versioning this indeed, as it seemed simpler to me. :sweat_smile:  I guess experience will tell.

maxwxyz, 2024-02-25:
Is this related to https://github.com/FreeCAD/FreeCAD/issues/5585 ?

maxwxyz, 2024-06-15:
Duplicate of #11936

closed 2024-06-15 by maxwxyz