FreeCAD/FreeCAD #5586: Provide ability to fine-tune auto constraints.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1560

* **Reporter:** CABrouwers
* **Date submitted:** 5/20/2014
* **FreeCAD version:** 0.13
* **Category:** Feature
* **Status:** assigned
* **Tags:** constraints, Solver

# Original report text

It could be nice to be able to select which constraints are automatically added when using auto constraints. For example, I usually would set the coincidence constraint on but I would rather turn the tangent constraint off because I find it is often creates unwanted constraints.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-05-22 13:32
Abdullah do you mind weighing in on this ticket?

---

### Comment by abdullah 2017-05-22 21:47
It is a rightful request. I also find tangency is sometimes getting in the way. It is just a new feature request... I assign it to myself ;)

adrianinsaval, 2023-03-28:
this could be implemented as a sub menu in the settings button of the constraint list

Jolbas, 2023-03-30:
Maybe while holding down mouse button there would be possible to toggle auto constraints by pressing corresponding shortcut key. H, V, O, C, N or T

abdullahtahiriyo, 2023-05-01:
The idea is not bad, but there should be some kind of indication of what is being activated/deactivated...

Jolbas, 2023-05-01:
I thought it could immediately snap to the nearest possible match and corresponding constraints mini icons is attached. But sometimes there may not be a nearest possible match and then I have no good idea. Maybe let the icons in the tool bar indicate by being temporarily greyed out or let the activated but not used mini icons hang around the pointer.

adrianinsaval, 2023-05-02:
the status bar could be used to display active auto constraints, we could even make toggles for them there

abdullahtahiriyo, 2023-05-03:
There is actually some possible interference. 

Shortcuts for constraint activation where thought to act while one DrawSketchHandler is active. So for example if one hits T to start a tangency constraint and realises it should be a perpendicular constraint, pressing N stops the DSH of the tangency and starts the one of the perpendicular.

I am not emotionally attached to one way or the other, but would need community discussion as it changes current behaviour.