FreeCAD/FreeCAD #5587: Request for Ability to Trim Surfaces in Part and Part Design WB's

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1567

* **Reporter:** bejant
* **Date submitted:** 5/30/2014
* **FreeCAD version:** trunk
* **Category:** Feature
* **Status:** assigned
* **Tags:** trim

# Original report text

The ability to trim a surface based on it's intersection with another surface would be a time-saver when modelling.  For example in the attached screen grab it would be handy to be able to trim off the upper part of the vertical surface above the horizontal surface in a point and click manner.

From the following forum thread:
http://forum.freecadweb.org/viewtopic.php?f=3&t=6706

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-02-09:
@donovaly does this ticket have anything to do with the work you've been doing on #8252 ?

donovaly, 2023-02-09:
> @donovaly does this ticket have anything to do with the work you've been doing on #8252 ?

No.

However this bug report can be closed. It does not provide sufficient info what exactly is missing. Since 2014 we have different tools to cut surfaces.

luzpaz, 2023-02-09:
An aside, anyone know bejant's github handle?

donovaly, 2023-02-09:
> @donovaly there is a forum thread with more details, did you check that before closing ticket?

I checked the linked forum thread. Last entry from 2014, no clear description what is missing and we have now the surface WB. We also have improved cut tools in Part WB.

If the problem persists, a new ticket with more details can be created.

In general, I tend to close issues when it is not 100 % clear what the case is. If there is still an issue, it is better to get a new one. A ticket that has not been touched for years will not be on the radar. Also it cost a lost of time for the devs to check what an old ticket might be about, if still valid etc. I therefore like the bug janitor in Bugzilla that closes such bugs automatically after some time. It issues you a warning that it will be closed, once not re-confirmed. If nobody reacts it is closed. I think we need such feature also for us.

luzpaz, 2023-02-09:
The point of migrating to github was to also review open tickets. And there hasn't been a systematic review of open tickets by the devs (only during releases and we've had just one release since). Beside me there have been only a few folk that have taken deep dives in to the issue queue.  IMHO, just closing tickets because their old doesn't make sense. It's like "throwing the baby out with the bath water".

wwmayer, 2023-02-09:
I disagree. The request is quite clear to me and currently not or not easily doable in Part or Surface wb.

Consider two intersecting faces or shells A and B where you want to trim face/shell A with face/shell B. The function is supposed to allow the user to click the part of the face/shell of A that should be kept.

donovaly, 2023-02-09:
> Beside me there have been only a few folk that have taken deep dives in to the issue queue. IMHO, just closing tickets because their old doesn't make sense. It's like "throwing the baby out with the bath water".

Oh, I appreciate your outstanding work!
It is that we have so many open feature requests that nobody looks at if they are still valid. Also note that on other projects, it is common to close feature requests as "WONTFIX", meaning nobody has time, too complicated etc. I have been part in different OpenSource development projects and in my opinion out of my experience the janitor service that closes feature requests after some time, is the best I found so far.
Note that this applies only for feature requests, not bugs. Verified bugs must stay of course open.
Also, this said, if  of course by subjective opinion.

Regarding this issue, i looked at the forum thread and to me it was not clear what exactly is missing that could not  be done using existing tools. (my approach in mind was to either bad to the desired surface then cut or use the surface WB to match a free-form surface.

donovaly, 2023-02-09:
> Consider two intersecting faces or shells A and B where you want to trim face/shell A with face/shell B. The function is supposed to allow the user to click the part of the face/shell of A that should be kept.

OK.
However, for me it is not clear how this feature could be implemented. I mean using only the Part WB allows you to cut the surface, here what I mean:
[test.zip](https://github.com/FreeCAD/FreeCAD/files/10701267/test.zip)

Maybe I misunderstood the report.

PLChris, 2023-02-10:
Sketcher has a trim tool which removes a part of a geometric element that is crossed by some other geometric element.
If we replace "geometric element" by "surface" we get what such a tool should do. In 3D things are a bit more complicated, because we need as a side condition that the surface to be trimmed is indeed completele traversed by the other surface.

Reusing donovaly's example, I think this should happen: Clicking on the green part of the vertical surface ...
[image]
... removes it and leaves you with this:
[image]
It could well be implemented on top of slice apart, which I used for the example shown here.

PLChris, 2023-02-10:
Moving discussions to Github was said to have the advantage of not having linear discussions only. How can that be done with the off topic "what to close" discussion here?

maxwxyz, 2024-03-02:
Demo of split with surface in Part Design on Link branch: https://youtu.be/qPoFrgk8S2U?si=uDGPZ87rcSlP_Lyr&t=1320

maxwxyz, 2024-08-30:
Is there any news on this issue? It is implemented in the Link branch @realthunder maybe you can weigh in.
@wwmayer you are currently assigned to this issue, is this still valid or do you want me to unassign you?

GAZ082, 2026-09-10:
This feature request is dead, looks like nobody is working on this. Also, looks like its parent is not in the 26.3 milestone.