FreeCAD/FreeCAD #5588: Part sweep bug with acute angle (Upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1570

* **Reporter:** denem
* **Date submitted:** 6/2/2014
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, OCC 7.7, OCC Bug, sweep, upstream

# Original report text

Sweep resulting figure contains invalid additional segments and/or obstructed parts. This bug affects FreeCAD .13 and .14

# Additional information

Sample project is attached along with screenshot of result.

# Steps to reproduce

Make any sweep containing an acute corner in wire.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: All All**
* **Platform:** All
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by shoogen 2014-06-02 12:54
reproduceable with OCCT development version:
> mksweep Sweep-0-spine
> setsweep -CF
> addsweep Sweep-0-section-00-Sketch001 
> buildsweep Sweep -C -S

Compile with option DEB : if problems with intersection : IntWalk_PWalking_1.gxx (lbr le 1erdec98)
Fail of TrimCorner

Draw[5]> checkshape Sweep
On Shape faulty_1 :
BRepCheck_NotConnected
Shape faulty_1  on shape faulty_2 :
BRepCheck_NotConnected

Faulty shapes in variables faulty_1 to faulty_2

---

### Comment by shoogen 2014-06-05 21:31
http://tracker.dev.opencascade.org/view.php?id=24991

---

### Comment by Kunda1 2017-01-14 21:30
upstream bumped ticket to OCC 7.2 
I've tagged this ticket accordingly

---

### Comment by Kunda1 2017-07-22 12:05
upstream bumped to OCC 7.2.1

---

### Comment by Kunda1 2017-07-22 12:08
Linking to #1571
denem can you re-up zips for both tickets ?

---

### Comment by Kunda1 2018-03-03 10:55
upstream bumped to OCC 7.4

---

### Comment by Kunda1 2019-09-01 19:40
Upstream bumped this to OCC7.5
tagged this ticket accordingly
updated OCC buglist https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

---

### Comment by Kunda1 2020-10-08 13:54
Upstream ticket was was bumped to OCC 7.6 roadmap.
Tagged this ticket OCC7.6
Notated in OCC bugtracker list: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

grubertechnic, 2023-08-21:
How could we test it, if there is no way to reproduce it. Even it is in the OCCT tracker, it does not mean, that the issue is valid. There were issues already in their tracker, where the issue was not valid.

luzpaz, 2023-09-24:
Upstream just de-prioritized this issue. They labeled it 'Unscheduled'. Doesn't look like it will get attention anytime soon.

[image]

grubertechnic, 2023-09-25:
IMHO, this issue can be closed, since it can not reproduced in any way, at least i did not find a anything useful in the links.

maxwxyz, 2024-02-25:
Please comment if this is still relevant, otherwise I will close it in a few weeks.

luzpaz, 2024-02-29:
Should still be relevant. Should not be closed unless unaminous community agreement (since this OCC related). This should also be applied to other longstanding OCC issues on the bugtracker.