FreeCAD/FreeCAD #5590: Points, curves and surfaces

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1589

* **Reporter:** Pauvres_honteux
* **Date submitted:** 6/23/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** feedback
* **Tags:** 

# Original report text

I'd like to propose a workbench for points, curves and surfaces similar to (preferably better than) what you have in the big french program.

# Additional information

With such a wb one can create anything, there are no limits what so ever once you got that. Ok, time maybe.
By the way, anyone here using that big french program at work?

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-09-27 17:19
Does OP mean something like NURBS ? 
ping chrisb

---

### Comment by Kunda1 2018-10-08 14:15
Pauvres_honteux please respond

---

### Comment by Pauvres_honteux 2018-10-09 05:27
Hi Kunda, I am having a **complete** surface workbench in mind. Nurbs would be a subpart of that workbench. Same goes for points and lines/curves.
As of now it's not possible since all datums are forced "behind bars", i.e. forced to be children of the Body container. (Datums being the absolute foundation of/for everything else)
So I guess we have to wait for that be fixed first. Hopefully it's not a big thing to fix?

---

### Comment by Kunda1 2018-10-09 12:15
Pauvres_honteux no idea. Please start a thread on the forum about this to deepen the discussion. Please link this ticket to it and vice versa. Thanks

luzpaz, 2022-02-07:
We have:  
* [Curves WB](https://github.com/tomate44/CurvesWB/) 
* [Points WB](https://wiki.freecad.org/Points_Workbench)
* [Surface WB](https://wiki.freecad.org/Surface_Workbench)

adrianinsaval, 2022-02-11:
@luzpaz reason for reopen? no mention of datum stuff on this ticket + no forum discussion

closed 2022-02-13 by luzpaz