FreeCAD/FreeCAD #5591: TCollection_IndexedDataMap::FindFromKey error when sweeping along BezierCurve (upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1646

* **Reporter:** ngallaher
* **Date submitted:** 7/29/2014
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, bezier curve, OCC 7.7, OCC Bug, sweep, upstream

# Original report text

Attempting to sweep a circle along a given BezierCurve to create a solid results in an error popup stating "TCollection_IndexedDataMap::FindFromKey"

# Additional information

OS: Ubuntu 12.04.4 LTS
Word size: 64-bit
Version: 0.15.3781 (Git)
Branch: master
Hash: 996af8d808a818726bd66c0e515e2d7eb7306bad
Python version: 2.7.3
Qt version: 4.8.1
Coin version: 3.1.3
SoQt version: 1.5.0
OCC version: 6.7.0

# Steps to reproduce

1) Open the reproduction file (unsweepable.fcstd).
2) Open the Part workbench
3) Click the "Utility to Sweep" tool
4) Select the circle and add it to the Sweep column
5) Click "Sweep Path".
6) Click on the BezierCurve to select it.
7) Click "Done"
8) Check "Create Solid"
9) Click "Ok"

Observe a message box titled "Input Error", containing the message TCollection_IndexedDataMap::FindFromKey

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Linux Ubuntu 12.04.4**
* **Platform:** x86_64
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by shoogen 2014-07-29 20:53
#5  0x00007f955ec02fd9 in TopTools_IndexedDataMapOfShapeListOfShape::FindFromKey (this=0x7fff41394450, K1=...)
    at occt/inc/TCollection_IndexedDataMap.gxx:354
#6  0x00007f955f6404c6 in IsSameOriented (aFace=..., aShell=...)
    at occt/src/BRepFill/BRepFill_PipeShell.cxx:198
#7  0x00007f955f643637 in BRepFill_PipeShell::MakeSolid (this=0x65b3440)
    at occt/src/BRepFill/BRepFill_PipeShell.cxx:840
#8  0x00007f955d4c98c8 in BRepOffsetAPI_MakePipeShell::MakeSolid (this=
    0x7fff41394720)
    at occt/src/BRepOffsetAPI/BRepOffsetAPI_MakePipeShell.cxx:274
#9  0x00007f95624d2c64 in Part::Sweep::execute (this=0x64beb60)
    at free-cad/src/Mod/Part/App/PartFeatures.cpp:463

---

### Comment by shoogen 2014-07-29 20:59
Error 3d = 24188.4831432816
30 Segment(s) of degree 11
Sweep : Degenerated case
An exception was caught 0x7f045d441b27 : Standard_OutOfRange: TCollection_IndexedDataMap::FindFromKey
** Exception ** 0x7f045d441b27 : Standard_OutOfRange: TCollection_IndexedDataMap::FindFromKey

---

### Comment by shoogen 2014-07-29 21:08
reported upstream http://tracker.dev.opencascade.org/view.php?id=25110

---

### Comment by Kunda1 2017-01-12 12:09
upstream set target to 0CCT7.2 
tagged this ticket accordingly

---

### Comment by wmayer 2017-09-01 18:34
With
OS: Ubuntu 14.04.5 LTS
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.17.11949 (Git)
Build type: Unknown
Branch: master
Hash: 5d2b22ca286207775ddcbeabef7c5850fda32203
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: German/Germany (de_DE)

it shows a different error message: BRepFill_Sweep::BuildEdge

---

### Comment by Kunda1 2018-03-03 10:58
upstream set target to 0CCT7.4
tagged this ticket accordingly
updated OCC buglist https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

---

### Comment by Kunda1 2019-09-01 19:36
Upstream bumped this to OCC7.5
tagged this ticket accordingly
updated OCC buglist https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

---

### Comment by Kunda1 2020-10-08 13:53
Upstream ticket was was bumped to OCC 7.6 roadmap.
Tagged this ticket OCC7.6
Notated in OCC bugtracker list: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

tgani, 2022-03-29:
I see the same problem when trying to create threads on parts. Please consider upping the priority on this as it's blocking a critical functionality.

tgani, 2022-03-29:
I have a similar issue when trying to create external threads with additive pipe in the below project:

[Fail_to_Create_Additive_Pipe.zip](https://github.com/FreeCAD/FreeCAD/files/8373204/Fail_to_Create_Additive_Pipe.zip)

Steps to reproduce:

1. Switch to Part Design
2. Make "Retaining ring" the active body
3. Click on "Sketch003"
4. Choose Additive Pipe from the Part Design tool menu
5. For "Object" click on the Helix001 displayed (or manually enter in dialog box)
6. Choose Selection orientation/Orientation mode as "Frenet"
7. Click OK and choose any option from the dialog box ("make independent copy etc.)

**Expected**: external threads using additive pipe are created
**Actual**: Dialog box pops up saying: NCollection_IndexedDataMap::FindFromKey and operation fails

_Sometimes_ this succeeds in creating threads but I have not be able to find out what causes it to succeed.

maxwxyz, 2024-02-25:
[image]
This is the current error in 0.22dev