FreeCAD/FreeCAD #5593: Allow Wrapping of 2D Objects, Draft ShapeString, Onto Non-Linear Surfaces

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1699

* **Reporter:** bejant
* **Date submitted:** 8/19/2014
* **FreeCAD version:** 0.14
* **Category:** Feature
* **Status:** new
* **Tags:** shapestring

# Original report text

It seems that users often want the ability to wrap 2D objects and text so it follows a curved surface.  One such forum post is here:
http://forum.freecadweb.org/viewtopic.php?f=3&t=7384

It could also be helpful if the user could select a + or - distance from the 3D face for the wrap.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by normandc 2014-08-19 03:01
Wrapping 2D geometry on surfaces would be useful for more than text. It is a standard surfacing tool.

---

### Comment by yorik 2014-08-19 15:40
But I'm afraid it's not very easy to achieve...

---

### Comment by bejant 2014-08-19 18:50
Sorry, but I goofed.  I mentioned 2D objects in the title but only mentioned text in the description.  Also, the last sentence in the description should read:

"It could also be helpful if the user could select a + or - distance FROM the 3D face for the wrap."

---

### Comment by Kunda1 2017-03-29 18:13
Linked to #897

Roy-043, 2023-07-15:
There is [Curves_SketchOnSurface](https://wiki.freecad.org/Curves_SketchOnSurface) now.

EmperorArthur, 2024-01-25:
Part_ProjectionOnSurface does 90% of this when the camera is looking directly at the surface.

I believe implementing #12116 would solve this issue.