FreeCAD/FreeCAD #5595: Add Reverse Fillet Tool or Option in Part / Part Design

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1742

* **Reporter:** quick61
* **Date submitted:** 9/11/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** GUI, Part, Part Design

# Original report text

The discussion has come up several times on the forum about the lack of a reverse fillet tool. The most recent one starting here - http://forum.freecadweb.org/viewtopic.php?f=3&t=7633&p=62333#p62329

Attached is an image of what the end result of the reverse fillet would be. If OCE will allow the implantation of such a tool, it would be a nice feature to have in the next release.  

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by quick61 2014-09-13 03:39
An older Forum discussion on this subject is here - http://forum.freecadweb.org/viewtopic.php?f=3&t=6667&hilit=ogee#p53838

And from this post - http://forum.freecadweb.org/viewtopic.php?f=3&t=6667&hilit=ogee&start=10#p53899 - it would seem that a proper name for the tool would be Cove.

---

### Comment by PedjaS 2020-11-02 20:17
This seems to be forgotten?

adrianinsaval, 2023-03-22:
I suggest closing this issue due to inactivity (since 2014) and unfeasibility, this functionality is not available in occt and implementing solely in freecad using sweeps or similar would likely be very fragile.

adrianinsaval, 2023-04-14:
I'll close this issue due to inactivity and being very niche, to interested parties if you really need such functionality _very often_ consider implementing this as a macro or addon. While this is not implemented in opencascade it would be very hard to implement it as a variant of the fillet functionality in FreeCAD, instead this type of thing has to be created as substractive pipes/lofts or pockets.

closed 2023-04-14 by adrianinsaval