FreeCAD/FreeCAD #5601: Add Devil (rounded, angled and tapered contact surface with a round hole in it)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1802

* **Reporter:** Pauvres_honteux
* **Date submitted:** 10/26/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** fillet, variable fillet

# Original report text

I'd like to propose a feature that takes away the effort from the user to make a so called "Devil". It's the rounded, angled and tapered contact surface with a round hole in it:
http://1.bp.blogspot.com/-xV7VHydI7F0/U52tHzHMg3I/AAAAAAAAdAY/GprPZqbatiY/s1600/1.jpg

With a bit of good will you may see it here too:
http://www.plmmarketplace.com/blog/content/binary/actas-mpa.png

# Additional information

I may make one at work for a really good close up picture to attach. 

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by yorik 2014-10-26 15:31
It's interesting, but I think you should describe much better all the particularities, properties and behaviours such feature should have, to interest a developer in coding it.

---

### Comment by Pauvres_honteux 2014-10-26 16:18
Hi Yorik, I'll do my best. Just hang in there!  =O)

---

### Comment by Pauvres_honteux 2014-11-02 20:22
Now, I tried it in FC and failed. So I did it at work in the cognac soaked programme.
I tried to make one picture per step as I went along, but I'm pretty sure I missed some steps here and there (my eyelids were almost touching my knees...).

The tricky part comes when the two cones radii are touching each other. I had to make radii without the cutting operation (as one normally does) in one step. When all fillets / radii where made I had to cut all surfaces and then join them. If one makes different radii on all surfaces it's no problem, but all radii equal spells trouble.

---

### Comment by Pauvres_honteux 2014-11-02 20:24
First of all: note that everything is done oversized, one shall always make sure all constituent parts, with guarantee, cross each other, no matter how much you change your parts later on.

The first 7 pictures just shows how I build up the double curved surface, including a bunch of intersections for easier visual orientation.

Picture 8 -> 13 shows how to make a common point, projection line, intersection point, planes for placing sketches on and a contact surface width common point.

Picture 14 -> 17 show how I build up the first cone including the projection of "the common point for surface width" and constrain it to the profile to revolve. Then cone 1 is split by the common plane.

Picture 18 -> 20 shows the second cone, which is made in the same way as the first one.

Picture 21 -> 23 shows the making of the "real" contact surface.

Picture 24 -> 26 shows the making of the hole cutting feature.

Picture 27 -> 30 shows the making of a "loose" fillet (Trim support 1 [and 2] are deactivated) so one later can make boundaries and later still extract one of them for the purpose of cutting a hole in the double curved surface.

Picture 31 -> 38 shows the making of the second cone and its constituent contact surface with related radius and boundaries.

Picture 39 shows the making of the "real connected" radius between first cone and double curved surface by joining them.

Picture 40 -> 46 shows the making of the second cones radius with the same technique as before.

Picture 47 -> 52 shows gradual visual transition to ready surface with everything tangent.

Picture 53 -> 56 shows the tangency problem areas.

Picture 57 -> 60 shows the power of this modelling technique.

---

### Comment by Pauvres_honteux 2014-11-02 20:36
I have in mind a function that takes one point on a surface as parameters, i.e. point and surface.
And then asks for the contact surface radius/diameter and plane.
And yet again for draft angles and radii.

Hope this straightened some question marks?

---

### Comment by Kunda1 2017-01-11 18:27
Updated title of ticket for better comprehension

---

### Comment by Kunda1 2017-05-12 12:29
yorik can you weigh in on this ticket?

---

### Comment by yorik 2017-05-18 02:11
Well you can probably already do something like that in FreeCAD with surfaces, but there could be a tool or macro to automate it, if someone wants to code it.

---

### Comment by Kunda1 2017-05-18 02:42
mario52 interested? Or perhaps propose this as a macro writing challenge?

---

### Comment by Kunda1 2017-06-24 11:09
Pauvres_honteux we lost the attachments to this ticket as well :(
Care to recreate?

---

### Comment by Pauvres_honteux 2017-06-24 13:04
Kunda1 , 
I'll check mi dungeon, I might have some luck...

---

### Comment by Kunda1 2017-06-26 12:41
Pauvres_honteux Another way to get this done is to build this as a macro yourself and then add to the FreeCAD-macro repo for others to download. What do you think?

---

### Comment by Pauvres_honteux 2017-06-27 05:06
Kunda1, as of now, to me it seems as the compressed tape archives I originally posted are back and working!
If they go away again, gimme a ping! =O)

---

### Comment by Pauvres_honteux 2017-06-27 05:10
Hmm, macro ye say? Mi n' macros... Let me put it this way, we know where we have each other...

luzpaz, 2022-06-02:
This ticket needs to be re-examined. None of the attached files in the original ticket are available anymore.

adrianinsaval, 2023-03-20:
it's from 2014, I think it qualifies as unreasonable amount of time open without action, I suggest closing, seems very niche too

closed 2023-03-20 by luzpaz