FreeCAD/FreeCAD #5603: Working plane not reset under Arch when making new project

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1823

* **Reporter:** lanciaux
* **Date submitted:** 11/10/2014
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** python

# Original report text

Current working plane settings are not reset after you create a new Project/Sheet.

# Steps to reproduce

1. Make anything you want
2. Go to Arch
3. Setup a plane setting by selecting any face of your model
4. Close this Project/Sheet without saving it
5. Create a new Project/Sheet

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **Updated:** 9/21/2021

# Discussion from Mantis ticket

### Comment by yorik 2014-11-10 23:55
Hm these are two independent things... I'm not sure if it would be a good thing or not to reset the work plane when opening a new model. You might have set it a certain way in another file, that might be convenient to keep it.

Besides, at the moment there is no mechanism to detect when a new document is opened from python, so this feature cannot be imlemented right now.

But let's leave it open, an idea could arise.

---

### Comment by Kunda1 2017-06-03 15:47
yorik can you weigh in on this again?

---

### Comment by yorik 2017-06-05 15:46
Still no idea on how to detect when a new document has been created/opened :(

---

### Comment by Kunda1 2017-10-18 11:03
[quote=yorik]Besides, at the moment there is no mechanism to detect when a new document is opened from python, so this feature cannot be imlemented right now.
[/quote]
wmayer care to weigh in?

---

### Comment by yorik 2018-12-05 12:58
This is now possible with the DocumentObserver. But since it can be disruptive, it's best to implement after the 0.18 release.

---

### Comment by Kunda1 2020-12-08 01:41
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=23&t=53021

---

### Comment by yorik 2021-09-21 08:56
***New proposal:*** implementing per-document working plane

*  The grid system is already per view, so each view can display a different grid already.
*  The python interface should stays the same. You just get and set the working plane and shouldn't worry about anything else
*  Two possibilities: 1) FreeCAD.DraftWorkingPlane still give you the current working plane of the document. On switching documents, this value gets changed. A doc observer changes the value each time the active doc is changed or 2) FreeCAD.DraftWorkingPlane is replaced with Draft.getWorkingPlane() which does all the work. I favour 2) but it will break some workbenches
*  The working plane data can be stringified and saved to document meta
*  Draft.getWorkingPlane() always reads from the document meta

yorikvanhavre, 2023-02-22:
@Roy-043 I'd like to start working on a per-document working plane system.

Basically I suggest this:

Fist step:

1. We implement Draft.getWorkingPlane() and Draft.setWorkingPlane() which does the same as FreeCAD.DraftWorkingPlane currently
2. Make sure all references to FreeCAD.DraftWorkingPlane are changed to Draft.getWorkingPlane()
3. We retire FreeCAD.DraftWorkingPlane to make sure nobody else uses it anymore

Second step:

1. We need a way to update the grid on FreeCAD active view change. Maybe just a small observer? Not sure it has a slot for that yet, need to investigate more... Maybe we can just hook to a built-in Qt signal of the MDI view...
2. Make Draft.setWorkingPlane() write the working plane to document.Meta
3. Make Draft.getWorkingPlane() read the working plane from document.Meta

And I think we're done...

Does that seem good to you?

Roy-043, 2023-02-22:
Related Forum topic:
[Draft Grid: improvements](https://forum.freecad.org/viewtopic.php?t=71565)
The code attached to the topic implements a (sort of) 'view observer' proposed here:
https://forum.freecadweb.org/viewtopic.php?t=26277&start=10#p566416

Your suggestions seem OK. Some remarks:

1. Using a function to get the current working plane is a good idea. It will allow us to later switch to a per-view working plane more easily should we wish to do so.
2. According to coding_conventions.md the names you propose are not correct. It's a detail, but still important enough to address at the beginning. The functions should be named `Draft.get_working_plane()` and `Draft.set_working_plane()` then.
3. Maybe both functions need optional arguments: `Draft.get_working_plane(doc=None)` and `Draft.set_working_plane(doc=None, plane=None)`.
4. I suppose we also need to store grid settings in the document meta data.
5. How to deal with the related preferences and the [Draft_SelectPlane](https://wiki.freecad.org/Draft_SelectPlane) command in the new situation? Add "Save as defaults", "Get defaults", "Apply to current document" and "Apply to all documents" buttons to the Draft_SelectPlane task panel and use the values from the preferences for new documents?
6. The gridSpacing is saved with units (as a string) in the preferences. When storing this value in the document meta data it may be better to use a float value instead. I personally would prefer gridSpacing to also be a float in the preferences BTW, and just add units in the GUI.
7. The Arch and BIM workbenches use `FreeCAD.DraftWorkingPlane` obviously, But some external workbenches also do (in the addons I have installed I can find references in Design456, dodo and freecad.trails). There may also be macros that rely on the old implementation. How should we deal with that?

Let me know if I can assist in any way.

yorikvanhavre, 2023-02-23:
Ok, yes of course agree with the set_working_plane notation. Old habits... :sweat_smile: 

```
from PySide2 import QtWidgets

def myFunction(args):
  print (args)

mw = Gui.getMainWindow()
mdi = mw.findChild(QtWidgets.QMdiArea)
mdi.subWindowActivated.connect(myFunction)
```
Yes, that's the best path forward IMHO. I'll have a look at your code!

Roy-043, 2023-02-23:
The `plane` argument will most likely not be optional for `Draft.set_working_plane()`.

This would be typical usage in the new situation?:
```
wp = Draft.get_working_plane()
wp.setTop()
Draft.set_working_plane(plane=wp)
```

yorikvanhavre, 2023-02-24:
Yes, that's what I have in mind too.
I also imagine:

Now:

```
wp = Draft.get_working_plane()
wp.setTop()
Draft.set_working_plane(wp)
```

Later:
```
wp = Draft.get_working_plane(doc=myDoc) # or wp = Draft.get_working_plane(view=myView)
wp.setTop()
Draft.set_working_plane(wp, doc=myDoc) # or Draft.set_working_plane(wp, view=myView)
```
And if the user does not specify doc or view, ActiveDocument/ActiveView are used, so it will still keep working like now.
Now, another  point: What should be saved to the document.Meta?
I think:
- wp
- grid settings

So no per-view settings would be saved. Which makes sense, because when you open a document anyway, only the last active view is restored. But then the question: Which view-dependent WP is to be saved to the doc.Meta? The last one?

So we actually need two structures:
- per-view wp + grid (kept for ex. in the Snapper object). That's where Draft.get_working_plane() will look to
- per document wp + grid. That's where Draft.get_working_plane() will look on file restore.
- Draft.set_working_plane() will always save to both, so the per document one always contains the last one used

How does that sound?

yorikvanhavre, 2023-02-24:
Additional idea: Extend the wp object so it also supports grid settings. So there is no more difference between the grid and the wp. The grid would basically become the "gui part" of the wp

Roy-043, 2023-03-17:
Sorry for the delayed response. Haven't had time to dive into this.

> The grid would basically become the "gui part" of the wp

That would be a good idea.

yorikvanhavre, 2023-03-20:
> Haven't had time to dive into this
Me neither :sweat_smile: 

It's done when it's done, FreeCAD motto ;)

Roy-043, 2023-09-01:
I have worked on this issue during the feature freeze. One thing that I think is required is to have multiple Plane classes. Since the current Plane class has some issues (old-style naming, some functions with confusing names, some duplicate code) we may need 2 additional Plane classes:

PlaneBase:
Parent class without any GUI-related attributes/methods.
This can be used when plane functionality is required independent from any document or view.

PlaneGui:
Based on PlaneBase with GUI-related attributes/methods:
- Grid settings.
- Icon, label and tooltip settings for the Draft Tray.

Plane/plane:
The old class kept for compatibility reasons. Where possible the old class should inherit from PlaneBase. It may prove to be hard to remove this class in the future.

In connection with this proposal I think some Plane related code should be moved to separate functions so that they can be accessed without creating a temporary plane object. The `placement_from_face` function (#10327) and `placement_from_points` function (#10296) are examples that have already been merged. But, for example, the `projectPoint` code should also be moved to a separate function IMO.

Do you agree with this proposal?

closed 2023-11-09 by Roy-043

luzpaz, 2023-11-09:
Thanks @Roy-043 !