FreeCAD/FreeCAD #5607: Integrate a client for the buildingSMART Data Dictionary web service

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1844

* **Reporter:** bofhjr
* **Date submitted:** 12/1/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** IFC

# Original report text

This would allow to  classify the different FreeCAD Arch model with the vocabulary and semantics of the IFCs --- It may also help for i18n

# Additional information

http://bsdd.buildingsmart.org
http://catenda.no/archives/2001

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

Moult, 2022-02-07:
Sure! https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.7.0/src/bsdd

Just a library for now, and I think the endpoints might've changed so I need to ask what the new ones were (they were migrating servers at the time we wrote the library).

luzpaz, 2022-06-02:
@Moult ping

Moult, 2022-06-13:
I have also really struggled to see a benefit of the bSDD as the database currently stands unfortunately. I guess people who use this nlsfb system will find it useful, but in Australia where it doesn't apply I have no idea. Maybe @berlotti knows the answer.

maxwxyz, 2024-02-25:
Is this issue still relevant as the discussion about the necessity was already two years ago without an outcome? Otherwise I will close this issue.

yorikvanhavre, 2024-02-26:
BSDD is still being developed by BuildingSMART and still encouraged... This mighty still interest people and motivate people to write some code for it. I think this could become a good GSoC idea. Let's leave this on, and I'll rewrite a bit the text.

berlotti, 2024-02-26:
Happy to contribute some sponsorship to this as well

Moult, 2024-02-26:
If it helps you can now mimic the interface built in BBIM.

RohanMishra315, 2024-04-02:
Hey @yorikvanhavre I have done research on it , how to implement this . So , after researching a lot and understanding the BSDD i  have a basic workflow how we can implement this. As always a lot of discussion with mentor is required.  I have made a proposal for Gsoc can you look at it and review it , thanks.

https://docs.google.com/document/d/1A2jbSJW3i4KBkCBKnlwhFTWN7xT3QWKOh2vHdqsau5w/edit?usp=sharing

drLite35, 2025-02-04:
Hi @yorikvanhavre @FreeCAD-Bug-Importer  , wanted to enquire if is this project there in GSoC'25 list and also wanted to enquire about https://wiki.freecad.org/Google_Summer_of_Code_2024 idealist if some information can be provided as well.

Also @yorikvanhavre https://developers.google.com/open-source/gsoc/timeline gsoc'25 timeline has been announced!!
Thanks!

maxwxyz, 2025-02-04:
generally we are open for all proposals. We have a `GSoC ` label on issues we think might be a good fit:
https://github.com/FreeCAD/FreeCAD/issues?q=state%3Aopen%20label%3A%22GSoC%22

yorikvanhavre, 2025-02-06:
We'll publish an updated GSoC page shortly!

semhustej, 2026-02-03:
bSDD is well and alive in 2026. 

I recorded a video on how to classify objects using bSDD data in Bonsai recently. You can see it here: https://youtu.be/Eqtqun9JrWM?si=IJROmQOeVqwvo64-

Moult, 2026-02-04:
Another video that might be good to watch is https://cloud.thinkmoult.com/ifc-videos/bsdd.mp4