FreeCAD/FreeCAD #5608: Extend Draft Trimex tool to change length property of basic Arch Structure object

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1866

* **Reporter:** rockn
* **Date submitted:** 12/19/2014
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

As a basic Arch Structure is a simple parametric box, could it be possible to extend the Draft Trimex tool to adjust the length of the Arch object depending of the selected face instead of create an extrusion ?

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: 14.04 64 bits**
* **Platform:** Linux Ubuntu
* **Updated:** 2/6/2021

luzpaz, 2023-10-24:
@yorikvanhavre any comments on this one ?

yorikvanhavre, 2023-10-27:
No, but I'll have a look if I can implement it!

blinhares, 2025-11-26:
If I understand correctly, is this the type of thing you're thinking of?
https://www.youtube.com/watch?v=pjzFwRR2TeE

closed 2026-09-12 by Roy-043