FreeCAD/FreeCAD #5609: Add MSAA 6x option

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1901

* **Reporter:** triplus
* **Date submitted:** 1/6/2015
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Older graphic hardware (Radeon R300g) supports up to MSAA 6x and having the possibility to select MSAA 6x under preferences would be great.

Side note: adding information to preferences GUI FreeCAD restart is needed when changing Anti-Aliasing mode probably wouldn’t hurt.

# Additional information

On Linux it is possible to force MSAA 6x mode for mentioned hardware like this:

export GALLIUM_MSAA=6
freecad

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Ubuntu 14.04**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by ickby 2015-01-08 06:49
Adding MSAA6x should be not a big issue, however, it may be a good idea to check which option is available in the hardware and only present those. I would need to investigate this.

The restart thing is only in current development version, 0.14 stable changes the AntiAliasing without restart. I have a few ideas how to change that, but that needs investigating.

I will takle both issues when my UI work is done, that may take a while

---

### Comment by wmayer 2015-01-08 10:46
ickby I hope you don't mind if I assign it to you then.

---

### Comment by ickby 2015-01-08 12:50
no, actually I wanted to do this too, somehow forgot that :)

---

### Comment by triplus 2015-01-08 18:29
If some (standard) way of checking hardware/driver available MSAA modes exist it for sure would be great for FreeCAD to only show available modes. For example as mentioned hardware does not support MSAA 8x setting this option produces the same result as if mode none was selected. No "radeon: Acquired access to AA optimizations." is printed to stdout.

User could get confused thinking MSAA does not work at all or thinking it might be FreeCAD bug. In reality hardware does not support MSAA 8x but MSAA modes up to 6x works just fine.

If there is no (standard) way of checking hardware/driver available MSAA modes exist i am guessing having 2x, 4x, (6x) and 8x will still work for majority of FreeCAD users just fine and don't spend to much time on this.

Changing Anti-Aliasing mode without restarting FreeCAD again that would be great and that should be priority when you decide to work on this again.

Cheers.

---

### Comment by wmayer 2017-09-14 13:44
http://blog.qt.io/blog/2015/09/21/using-modern-opengl-es-features-with-qopenglframebufferobject-in-qt-5-6/
https://stackoverflow.com/questions/24738638/how-to-determine-the-maximum-number-of-supported-multisample-samples
https://docs.blender.org/manual/ja/dev/preferences/system.html

luzpaz, 2022-05-13:
@wwmayer can you weigh-in on this?

wwmayer, 2022-05-14:
Here is a method to test the supported MSAA modes:
```
class GLFormatCheck {
public:
    GLFormatCheck() {
        context.setFormat(format);
        context.create();
        offscreen.setFormat(format);
        offscreen.create();
        context.makeCurrent(&offscreen);
    }

    bool testSamples(int num) {
        QOpenGLFramebufferObjectFormat fboFormat;
        fboFormat.setAttachment(QOpenGLFramebufferObject::Depth);
        fboFormat.setSamples(num);
        QOpenGLFramebufferObject fbo(100, 100, fboFormat);
        return fbo.format().samples() == num;
    }

private:
    QSurfaceFormat format;
    QOpenGLContext context;
    QOffscreenSurface offscreen;
};

void testModes()
{
    GLFormatCheck check;
    bool msaa2x = check.testSamples(2);
    bool msaa4x = check.testSamples(4);
    bool msaa6x = check.testSamples(6);
    bool msaa8x = check.testSamples(8);
}
```

On my system msaa2x, msaa4x and msaa8x are all true and msaa6x is false. But I think this should be implemented in v0.21 just in case the method doesn't work (e.g. crashes) on some systems.

closed 2024-03-25 by sliptonic