FreeCAD/FreeCAD #5612: Fuzzy Boolean Operations

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1967

* **Reporter:** shoogen
* **Date submitted:** 2/14/2015
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

make use of Fuzzy Boolean Operations
http://dev.opencascade.org/index.php?q=node/1056

# Other bug information

* **Priority:** none
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by shoogen 2015-03-22 22:21
git://github.com/5263/FreeCAD dev-fuzzyboolean

The development has stalled. Please feel free to test and comment. I would like to wait with the integration until the problem with corrupted input shapes (comment-520) is solved.

---

### Comment by openBrain 2019-08-31 10:57
As of today, fuzzy boolean operations are available in the Defeaturing WB.

luzpaz, 2022-07-10:
@0penBrain what do you think, should we close this ticket ? I just wonder how people will learn of defeaturing wb just using vanilla freecad...you know?

0penBrain, 2022-07-11:
@luzpaz have no clue about the ticket. Just depends if we want this to be implemented in FC or not. :) From a quick reading of the article, to me it looks like it could have some added value. ;)

maxwxyz, 2024-02-25:
The idea is to implement this into Part / PartDesign WB for boolean operations?

FEA-eng, 2025-06-04:
@FreeCAD/cad-working-group I think this would be really useful. Can you add it to your list?

FlachyJoe, 2025-06-04:
`TopoShape::makeElementBoolean` already has a `tolerance` argument (default to -1 so the AIRCAP auto fuzzy value is applied).
It's quite easy to add a property to objects which use it and change the calls.

FlachyJoe, 2025-06-04:
IMHO it is urgent to wait: #21713 and #21574 will make this easiest to implement.

maxwxyz, 2026-04-25:
The mentioned PR was merged