FreeCAD/FreeCAD #5614: Boolean Operations With Multiple Arguments

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=1971

* **Reporter:** shoogen
* **Date submitted:** 2/17/2015
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Use the Boolean Operations With Multiple Arguments for the MultiFusion and MultiCommon Featues.
Make them available from python.
http://dev.opencascade.org/index.php?q=node/1060

# Other bug information

* **Priority:** high
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by shoogen 2015-02-19 07:42
It seems that this is not suitable to be used for MutliCommon Features.
The Cut function might be useful in PartDesign array Features.

---

### Comment by shoogen 2015-03-21 22:12
http://tracker.dev.opencascade.org/view.php?id=25966 in response to http://forum.freecadweb.org/viewtopic.php?f=3&t=10225

luzpaz, 2022-07-11:
Upstream OCC ticket (http://tracker.dev.opencascade.org/view.php?id=25966) was fixed in OCC v7.3

Jolbas, 2023-03-30:
Isn't this resolved and can be closed? It's possible to add as many objects you like to the Shapes property of a Part::MultiFuse or a Part::MultiCommon. And the case in the forum thread does work as expected in current master

wwmayer, 2023-03-31:
The ticket is about using the methods BRepAlgoAPI_BooleanOperation::SetArguments and BRepAlgoAPI_BooleanOperation::SetTools because this way it's more efficient than performing several boolean operations in a for loop. For MultiFuse this is done but not yet for MultiCommon

closed 2024-03-14 by wwmayer