FreeCAD/FreeCAD #5619: Boolean ->  Difference leaves a extra hole (upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2024

* **Reporter:** bhs67
* **Date submitted:** 3/23/2015
* **FreeCAD version:** 0.14
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** #ready-for-migration, bump, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

One attachment is a pdf indicating the problem.  The other two attachments are before and after the Procedure.

# Steps to reproduce

Procedure ... Part -> Boolean -> Boolean -> Difference -> (first shape) cube -> (second shape) total hollow fillet -> Apply.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Windows 7 SP1**
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by shoogen 2015-03-23 17:01
You did not mention whether you created this file with automated refines enabled.
The total hollow fillet is invalid in both files.
You did not specify your full FreeCAD version information.

---

### Comment by bhs67 2015-03-23 17:14
I am using FreeCad 0.14 rev 3700.
Where do I find automated recomputes?

---

### Comment by shoogen 2015-03-23 22:30
sorry, i meant automated refines. The are enabled by a setting called "automatically refine shape after boolean operation"

---

### Comment by bhs67 2015-03-24 14:58
Hmmm.  I checked the Preferences ... do not have  "automatically refine model after boolean operation" checked. Should it be checked or not checked?  Thanks!

---

### Comment by shoogen 2015-03-24 17:37
http://tracker.dev.opencascade.org/view.php?id=25979

---

### Comment by Kunda1 2017-01-15 15:22
shoogen should we bump the upstream bug ?

---

### Comment by Kunda1 2017-08-27 12:38
ickby do you mind asking the upstream devs to assign this to their roadmap?

---

### Comment by ickby 2017-08-30 08:27
Kunda: the bug has been verified against current OCCT recently and a test case has been added: 2017-08-07 18:16

I think this should be enough to gurantee the treatment, what 
do you think?

---

### Comment by Kunda1 2017-08-30 18:32
ickby it's your call. I personally would ask if they could at least assign it their 'unscheduled' list.

luzpaz, 2022-05-26:
Upstream ticket still open http://tracker.dev.opencascade.org/view.php?id=25979

grubertechnic, 2023-07-04:
@luzpaz : see https://forum.freecad.org/viewtopic.php?t=20264&start=110

It is already solved (also @wwmayer mention it in the thread, that we probably can close it, even when it is upstream still open).

Here are some attachments about the invalid brep, a file to retest and a screenshot.

[Fillet001.brep.zip](https://github.com/FreeCAD/FreeCAD/files/11950794/Fillet001.brep.zip)
[issue6268.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11950801/issue6268.FCStd.zip)
[image]

closed 2023-08-24 by luzpaz