FreeCAD/FreeCAD #5622: FreeCAD warns when applying thickness "BRepAlgo_Image::Bind"

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2041

* **Reporter:** chrisb
* **Date submitted:** 4/7/2015
* **FreeCAD version:** 0.14
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, OCC Bug, upstream

# Original report text

FreeCAD crashes when applying thickness to the selected surface of a lofted part. 

<b>EDIT: Jan 23 2017</b>
Not crashing in 0.17. It shows an alert instead: <code>BRepAlgo_Image::Bind</code>

<b>EDIT: Jan 24 2017</b>
Other users are reporting that this is still a crash bug but are still using version that is inferior to OCC 7.0

<b>EDIT: Mar 25 2017</b>
Using freecad-daily with OCCT 7.1.0, no crash and no error: the Thickness completes, but the result is unchanged from the Loft (no shelling operation is done). Part Check geometry reports the solid as invalid.

# Additional information

Other members checked the error via the forum: http://forum.freecadweb.org/viewtopic.php?f=3&t=10382

Note: This ticket is being tracked via [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264) forum thread

# Steps to reproduce

Open the file uploaded to the forum: http://forum.freecadweb.org/download/file.php?id=12633 , select part workbench, select the surface and click the thickness icon.<code>OS: Ubuntu 16.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.10611 (Git)
Build type: None
Branch: master
Hash: 44a9ecd25f27ec3efe52c9de9c95ef2b3e19dd17
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0</code>

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 14.04**
* **Platform:** Linux, Mac, Windows
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by shoogen 2015-04-08 08:21
pload MODELING
restore thickness-crash1-tcl-Loft001-00-section.brep Loft001-00-section
restore thickness-crash1-tcl-Loft001-01-section.brep Loft001-01-section
thrusections Loft001 1 0 Loft001-00-section Loft001-01-section
explode Loft001 F
offsetparameter 1e-7 p a
offsetload Loft001 1 Loft001_2
offsetperform Thickness
donly Thickness

---

### Comment by Kunda1 2017-01-23 23:16
per chrisb in https://forum.freecadweb.org/viewtopic.php?f=3&t=10382&start=10#p155463
<code>Not crashing in 0.17. It shows an alert instead:
BRepAlgo_Image::Bind.</code>

OS: Mac OS X
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.9586 (Git)
Build type: Release
Branch: (detached from 0e2cf2e)
Hash: 0e2cf2ebcaaa50723de608cf030af740cb6ad548
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 7.0.0

---

### Comment by Kunda1 2017-03-25 11:38
Per normandc in https://forum.freecadweb.org/viewtopic.php?f=3&t=10382&start=10#p167101
[quote=normandc]Using freecad-daily with OCCT 7.1.0, no crash and no error: the Thickness completes, but the result is unchanged from the Loft (no shelling operation is done). Part Check geometry reports the solid as invalid.
[/quote]
<code>OS: Ubuntu 16.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.10611 (Git)
Build type: None
Branch: master
Hash: 44a9ecd25f27ec3efe52c9de9c95ef2b3e19dd17
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0</code>

---

### Comment by Kunda1 2017-12-06 22:27
wmayer please advise. Is this still an upstream OCC issue ?

---

### Comment by wmayer 2019-01-18 23:00
For me it works only when selecting Face5 (with OCCT 7.0 and 7.2). But when selecting any other face it fails and the error message is: BRepAlgo_Image::Bind

maxwxyz, 2026-01-07:
@pieterhijma FYI

pieterhijma, 2026-01-21:
The file: [errorThickness.FCStd.zip](https://github.com/user-attachments/files/24769471/errorThickness.FCStd.zip)

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.2.0dev.45085 (Git)
Build date: 2026/01/21 13:49:08
Build type: Unknown
Branch: main
Hash: 457a2eb300027df7019274d4c667cf79c088ce36
Python 3.14.2, Qt 6.10.1, Coin 4.0.6, Vtk 9.5.2, boost 1_89, Eigen3 5.0.1, PySide 6.10.1
shiboken 6.10.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
  * BCFPlugin 1.0.0
  * Ondsel-Lens 2025.11.21.01
```

- Face 1: works
- Face 2: works
- Face 3: works
- Face 4: does not work, no error message, geometry check: Face6: Invalid tolerance value
- Face 5: works
- Face 6: works

codeotaku, 2026-08-31:
Different combinations of Arc/Intersection seem to work:

Face1 (arc join type + intersection check box)
[image]

Face 2 (arc join) 
[image]

Face 3 (arc join)
[image]

Face4 (Intersection join type):
[image]

Face5 (arc join + intersection check box)
[image]

Face6 (arc join)
[image]

OCC 7.8.1

OS: macOS 26.6.2
Architecture: arm64
Version: 26.3.0dev.20260728 (Git shallow)
Build date: 2026/07/28 21:59:12
Build type: Release
Branch: (HEAD detached at 63af5cb2f)
Hash: 63af5cb2f3dbb2b4c7333f6d220b03e05bf92f99
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: cocoa/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
OpenGL version/vendor/renderer: 2.1 Metal - 90.5/Apple/Apple M3 Max
Installed mods:
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.74
  * Fasteners Workbench 0.5.64
  * PieMenu 1.13.0 (Disabled)
  * SheetMetal Workbench 0.8.21
  * Thread Workbench 0.3.1