FreeCAD/FreeCAD #5623: Add Chamfer Edge-Tool

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2052

* **Reporter:** r-frank
* **Date submitted:** 4/13/2015
* **FreeCAD version:** 0.14
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Like the already existing "Fillet Edge" tool to apply a fillet to two lines, there should be a tool to apply a chamfer as well. Could use the same logic (workflow) as the "Fillet Edge" Tool.
See discussion here:
http://forum.freecadweb.org/viewtopic.php?p=83906#p83906

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Windows 7 64 bit**
* **Platform:** Windows
* **Updated:** 2/6/2021

adrianinsaval, 2022-02-11:
@PaddleStroke has implemented the tool but it's not merged yet, I think it's mixed in a PR for another feature. Let's see how it goes.

maxwxyz, 2024-02-21:
@PaddleStroke I couldn't find the PR, just tag this issue if you submit.

PaddleStroke, 2024-02-26:
I don't have a PR for this yet

PR #12898 (2024-03-18): Sketcher: Chamfer tool (and fillet refactor)
- Rework fillet tool, so that preserve corner is a checkbox not a separate tool.
- Introduce Chamfer tool.
- Few syntax fixes/modernization in fillet()
- unrelated but fix error in className of CmdSketcherCompHorizontalVertical
[image]
Fixes #12464 (though there is no OVP introduced)
Fixes #5623
closed 2024-03-18 by WandererFan