FreeCAD/FreeCAD #5628: 3D printer simulator - a new nodekit?

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2091

* **Reporter:** philius
* **Date submitted:** 5/6/2015
* **FreeCAD version:** 0.15
* **Category:** General
* **Status:** new
* **Tags:** 

# Original report text

I tried understanding how the robots example works, but as a newbie the number of external references is daunting.

I want to create a 3D printer model in freecad that shows a delta printer, although any other printer would help, but I'm using delta printer terms below.

Its shows its print volume - the area it can 3D print in, in a different color.
There are a couple of sliders to change:
Delta arm length: affects print volume width+height
Linear actuator length: affects print height, required linear bearings diameter
Base radius: affects print area
Delta arm distance: affects actuactor dimensions (the actuator is connected by 3 arm-pairs) which affects the print volume width

It looks like the 6-axis robot model was developed outside of freecad, but it would be great if such models could be developed, edited and shared through freecad!

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** General
* **OS: Debian 8**
* **Platform:** amd64
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by philius 2015-05-07 11:12
Something like http://www.thinkyhead.com/_delta/
but with a translucent print volume with some sliders.

---

### Comment by Kunda1 2017-10-23 18:35
Interesting developments in the Path WB: https://forum.freecadweb.org/viewtopic.php?f=10&t=24923&p=196416#p196414

---

### Comment by abdullah 2017-12-19 13:36
kunda1

This is not sketcher related.

Do we need a robot WB project?

---

### Comment by Kunda1 2017-12-19 14:39
CC yorik

---

### Comment by yorik 2018-01-05 19:34
Okay I added a Robot subproject.

maxwxyz, 2024-02-25:
Please add feedback if this is still relevant.

maxwxyz, 2024-04-06:
I'll close this for no more feedback

closed 2024-04-06 by maxwxyz