FreeCAD/FreeCAD #5632: Crash when unioning toruses (upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2204

* **Reporter:** biogeo
* **Date submitted:** 7/30/2015
* **FreeCAD version:** 0.15
* **Category:** Bug
* **Status:** resolved
* **Tags:** #pending, #ready-for-migration, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

When using the part workbench to union two toruses, in certain situations FreeCAD will appear to enter an infinite loop, during which it gradually consumes more and more RAM until the system slows to a crawl and the only solution is to kill the process.

I have found this to occur under the following conditions:
1. For both toruses, Radius1 == Radius2, and they are also equal between the two toruses.
2. For both toruses, Angle3 == 90 degrees, i.e., they are both a "quarter donut".
3. The two toruses are positioned to share a circular face.

Depending on the relative angle between the planes in which the toruses lie, the union operation will sometimes work correctly, sometimes "succeed" but produce an unexpected result (e.g., only three circular faces with no torus components), and sometimes produce the infinite loop condition described above.

# Steps to reproduce

1. Open a new file and go to the Part workbench
2. Create a torus "Torus" with Radius1=2mm, Radius2=2mm, Angle3=90
3. Create a second torus "Torus001" with Radius1=2mm, Radius2=2mm, Angle3=90
4. Rotate Torus001 90 degrees about the z-axis. The two together should look like a single torus with Angle3=180.
5. Select Torus and Torus001 and use the Union tool.

FreeCAD enters an infinite loop gradually allocating more and more RAM until the system slows to a crawl and the process must be killed.

The attached file is the result of steps 1-4. Opening this file in a new FreeCAD session and running Union on the two toruses reliably crashes FreeCAD on my system.

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **OS: Linux Mint 13 Maya**
* **Updated:** 1/24/2022
* **Fixed in version:** 0.2

# Discussion from Mantis ticket

### Comment by biogeo 2015-07-30 20:25
Forgot to include as additional information:

OS: Linux Mint 13 Maya
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.3
Qt version: 4.8.1
Coin version: 3.1.3
OCC version: 6.8.0.oce-0.17

---

### Comment by wmayer 2016-02-28 18:32
That's clearly an OpenCascade bug which we are not able to fix.

---

### Comment by jmaustpc 2016-03-01 15:17
It eventually crashes FreeCAD here, with the OCE from tanderson, OCE patches over OCC 6.9.1

OS: Ubuntu 12.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6523 (Git)
Build type: Release
Branch: master
Hash: ecd6517cb00fcf05edd41a38b3ce1fa8b16ec3ad
Python version: 2.7.3
Qt version: 4.8.2
Coin version: 3.1.3
OCC version: 6.9.1.oce-0.18-dev

---

### Comment by Kunda1 2017-01-13 17:12
Does this issue persist in OCC 7.x ?
If so Is there an open upstream bug we can link to?

---

### Comment by Kunda1 2017-06-10 19:54
biogeo can you re-up the attachment? We restored recently from a backup after an issue we had with th tracker and some attachments like yours didn't make it.

---

### Comment by Kunda1 2017-08-21 11:01
jmaustpc do you mind testing this with OCC7.1 ?

---

### Comment by wmayer 2017-09-02 10:43
Attached is a file that shows the behaviour with occ6.8 where the applications eats all memory.
With occ7.2 it now returns immediately but the shape is invalid. On the other hand in practise such a shape doesn't make too much sense anyway.

---

### Comment by Kunda1 2018-05-30 16:05
Needs to be tested with OCC7.3

---

### Comment by Kunda1 2019-02-27 21:57
wmayer can you reproduce in OCC7.3

---

### Comment by wmayer 2019-02-28 10:22
Yes, the behaviour with OCC7.3 is the same as with 7.2

---

### Comment by wmayer 2021-12-12 14:44
With OCC7.4 and 7.5 still invalid shapes are created but with 7.6 finally a correct shape is built.

---

### Comment by wmayer 2021-12-12 14:45
Needs to be built with OCC 7.6

luzpaz, 2022-05-28:
The example file for this bug isn't available anymore.  
Should be tested against OCC v7.6.2

wwmayer, 2022-05-28:
[Torus-union.fcstd.zip](https://github.com/FreeCAD/FreeCAD/files/8791702/Torus-union.fcstd.zip)

AjinkyaDahale, 2023-02-25:
@luzpaz can skip forum discussion now I guess.

AjinkyaDahale, 2023-02-25:
No crash when making a union with

```
OS: Ubuntu Core 20 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32018 (Git) Snap 625
Build type: Release
Branch: master
Hash: 5f3e781dbf280c5f8c97183414e2aeb93a57b0be
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

luzpaz, 2023-02-26:
@wwmayer can you reproduce as well ?

wwmayer, 2023-02-26:
On my old system I had a self-compiled OCCT 7.6 and there it worked. See my comment from 2021-12-12 14:45

Jolbas, 2023-03-31:
No crash for me. Neither on:
```
OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.32630 (Git)
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
or
```
Version: 0.20.1.29410 (Git)
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
```

AjinkyaDahale, 2023-03-31:
This should be closed by now as either "cannot reproduce" or "resolved on updating dependency"

luzpaz, 2023-04-01:
Nice. Closing.

closed 2023-04-01 by luzpaz