FreeCAD/FreeCAD #5641: Support for IfcProfileDef

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2448

* **Reporter:** yorik
* **Date submitted:** 2/11/2016
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** IFC

# Original report text

Structural elements that are based on a profile should be translated to/from IFC as IfcProfileDef objects.

# Additional information

https://github.com/IndustryFoundationClasses/Questions/issues/3

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Debian Testing 64bit**
* **Platform:** PC
* **Updated:** 2/6/2021

luzpaz, 2023-03-18:
@yorikvanhavre is this relevant to core or should be in nativeIFC workbench ?

yorikvanhavre, 2023-03-20:
I think the problem could vanish with NativeIFC, but it still makes sense to keep it here for now I think, as it might be interesting at some point to specifically need to convert a non-NativeIFC FreeCAD 2D object to an IfcProfileDef...