FreeCAD/FreeCAD #5642: Crash when creating loft between two sketches

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2461

* **Reporter:** taiyo_uk
* **Date submitted:** 3/2/2016
* **FreeCAD version:** 0.15
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, OCC 7.3, OCC Bug, upstream

# Original report text

Crating a loft between two sketches causes FreeCAD to segfault in Fedora 23 x86_64, and to produce this error in Windows 7 64-bit: "ACCESS VIOLATION at address 0x00000010 during 'READ' operation"

This only happens for certain sketches such as those included in the attached file.

# Additional information

Changing the order of the two sketches has no effect. Selecting the closed option produces a different error in Windows: "TCollection_DataMap::ChangeFind"

# Steps to reproduce

1. Open the attached file in FreeCAD
2. Create a loft using the two sketches
3. The crash happens when clicking the OK button in the loft dialog.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows Windows 7 64 bit**
* **Platform:** x86_64
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by taiyo_uk 2016-03-02 11:20
Version information:

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

---

### Comment by taiyo_uk 2016-03-03 07:22
The same problem happens on the latest (at the time of writing) git build, this is what happens on Fedora:

*** Abort *** an exception was raised, but no catch was found.
	... The exception is:SIGSEGV 'segmentation violation' detected. Address 0

Version information:

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6538 (Git)
Build type: Release
Branch: master
Hash: 57c783e5b0f4da8611f5b1f26dc7499a5979e574
Python version: 2.7.10
Qt version: 4.8.7
Coin version: 3.1.3
[... 169 lines cut ...]
Locale: French/Canada (fr_CA)

---

### Comment by normandc 2019-02-09 20:21
No crash with OCC 7.1.0 with this version. An error message appears in a pop up when pressing OK:
[quote]BRepFill :: profiles are inconsistent[/quote]

OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13509 (Git) AppImage
Build type: None
Branch: master
Hash: 0258808ccb6ba3bd5ea9312f79cd023f1a8671b7
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: French/Canada (fr_CA)

---

### Comment by wmayer 2019-02-23 20:29
I don't have OCCT 7.3 to this but according to the call stack this anyway must be a bug in OCCT, then.

JanneKro, 2022-05-11:
I have tested with the example posted by "hirnsalat" in [https://tracker.freecadweb.org/view.php?id=2461#c11373](https://tracker.freecadweb.org/view.php?id=2461#c11373) with 4 splines, but did not manage to get it to crash. Tried adding/removing sections in different orders and it worked just fine. 

The original file is no longer available.  Close?

luzpaz, 2022-05-12:
Please always add the [full version info](https://wiki.freecad.org/About) to the post you've tested so we know what OS and specs/deps you've tested it against

JanneKro, 2022-05-12:
Tested without issues.
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28835 (Git)
Build type: Release
Branch: master
Hash: 55f1222bc29e5e2338ff86175175aa3118faeb01
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/Sweden (en_SE)
[/code]

Johnny846, 2022-07-17:
Tested, 
All (Part Design) loft combinations done with only 2 sketches had no issues, except these 2 combinations which produced a weird-looking loft when selected in the following order: 
-Sketch003 selected before Sketch001
-Sketch003 selected before Sketch004

All combinations done with 3 or 4 sketches caused my FreeCAD instance to close right after clicking the loft button, although I don't know if it's a segmentation fault, as the (journald?) log shows the same entries as if I normally close FreeCAD (using the cross button).

OS: Ubuntu 22.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * fcgear

AjinkyaDahale, 2022-09-30:
Segfault also observed with:
```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.30485 (Git)
Build type: Release
Branch: fem-rigid-body-constraints
Hash: 8924e265c068c2446a091f84c83c8b1934451470
Python 3.10.6, Qt 5.15.5, Coin 4.0.1, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.5.7
  * Curves.backup1662220878.4365516 0.4.4
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

Looks like an OCC issue from the stack trace.

Jolbas, 2023-03-31:
I get same result as Johnny846 with OCC 7.6.3. Crash when using 3 or 4 sketches but not when using 2 sketches.

```
OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.32630 (Git)
Build type: Release
Branch: ZeroLengthPD
Hash: d65b05c3df48c690bdcf7f9d982cb96dbe288911
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

AjinkyaDahale, 2023-03-31:
@luzpaz Do we have an OCC ticket?

luzpaz, 2023-04-01:
@AjinkyaDahale nope :cry: 
[image]  
(per https://forum.freecad.org/viewtopic.php?t=20264)

AIRCAP, 2024-09-26:
For the record, 

```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38842 (Git)
Build type: Debug
Branch: anotherLoftFix
Hash: 69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

still has this issue (not fixed in OCCT 7.6.3)

also #16853 does **NOT** fix this

leoheck, 2025-05-15:
Not Ruled loft with OCCT 7.8 does not Crash FC.

Could it be improved further with OCCT 8.0-RC1?

[image]

```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

leoheck, 2025-05-15:
But looking from the back, the shape does not look good.
It looks like there are some (black) edges missing on the right.

[image]

[image]

pieterhijma, 2026-10-27:
On FreeCAD main with OCCT 8.1 dev, this doesn't crash anymore. This has also been reported with other OCCT versions.

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

I'm removing the crash labels but the geometry that is created with this loft is faulty. I'm going to change the title of the issue.

luzpaz, 2026-10-28:
@pieterhijma is the remaining problem of this ticket still an OCC issue ? If so should the OCCT Liason status change too ?

Edit: nvm. I was confused. It was changed. Cancel clear.